veryl-lang/veryl #5: Loop statement support

taichi-ishitani, 2022-12-21:
My existing SV code includes loop statements such as for loop and while loop so I'd like Veryl to support these statements.

I think SV (C lang) style for loop statement is very bother notation because there are many similar code for example  `int i = 0;` and `++i;`. So I'd like you to introduce simple notation.

dalance, 2022-12-21:
I'll add Rust style `for` statement like below:

```
for i: u32 in 0..MAX {
}
```

This style can't specify the range step. ( In Rust, it can be `(0..MAX).step_by(2)` )
So the additional notation to specify step ( or other incrementation ) is needed. 

```
for i: u32 in 0..MAX += 2 {
}
```

This is just idea.

taichi-ishitani, 2022-12-22:
> for i: u32 in 0..MAX += 2

This notation looks like below.

```
for i: u32 in 0..(MAX += 2) 
```

So I think a separator symbol or keyword is needed.

```
for i: u32 in 0..MAX step_by 2
```

dalance, 2022-12-22:
The syntax of range will be `EXPRESSION .. EXPRESSION`,
and `+=` will be statement. so `(MAX += 2)` is invalid.

On the other hand, the same symbol may be confusable.
The following notation can specify more operation.

```
0..MAX step += 2
0..MAX step -= 2
0..MAX step *= 2
0..MAX step /= 2
```

taichi-ishitani, 2022-12-22:
> 0..MAX step += 2

This notation looks good for me.

dalance, 2022-12-22:
Now for/if statement and for/if declaration ( a.k.a generate for/if ) are supported.

closed 2022-12-25 by dalance