veryl-lang/veryl #11: case statement

dalance, 2022-12-22:
```
always_ff (clk) {
    case x {
        0: a = 1;
        1: a = 2;
        2: {
            a = 3;
            b = 4;
        }
        default: a = 4;
    }
}
```

taichi-ishitani, 2022-12-22:
Do you plan to add `case inside` feature?

dalance, 2022-12-22:
I want to consider it, but I think the priority is low.

dalance, 2022-12-29:
Instead of `default`, How about Rust style default?

```
always_ff (clk) {
    case x {
        0: a = 1;
        1: a = 2;
        2: {
            a = 3;
            b = 4;
        }
        _: a = 4;
    }
}
```

taichi-ishitani, 2023-01-05:
In SysemVerilog, `?` and `*` are used as symbols which mean wildcard. For exmaple: wildcard operator `==?`, wildcard index type `int a[*]`.
How about using `?` or `*` instead of `_`?

dalance, 2023-01-05:
I thought `default` may be a little long. but there is not another problem.
So I'll use `default` to follow SystemVerilog.

Do you have any other reason to change to `?` or `*`?

dalance, 2023-01-06:
OK. I'll use `default` at this time.

taichi-ishitani, 2023-01-11:
How about introducing case item less case statement.
Example:

```
case {
  foo:     a = 1;
  bar:     a = 2;
  default: a = 3;
}
```

Equivalent code:

```
case 1 {
  foo:     a = 1;
  bar:     a = 2;
  default: a = 3;
}
```

This idea is originally from Ruby's case expression.

[image]

dalance, 2023-01-11:
It looks good. Does Expression in case item have no problem for synthesis?

taichi-ishitani, 2023-01-11:
> Does Expression in case item have no problem for synthesis?

Yes. In fact, our RTL includes this kind of code.

dalance, 2023-01-12:
Expression less case seems to be difficult to parse by LL(k).
This is because

```
case {  <- Is this `{` case body's start or expression with concatenation?
    foo: a = 1;
}
```

To avoid this, chanigning `case` keyword for expression less case is required.

taichi-ishitani, 2023-01-12:
I don't care if expression less case feature is not introduced because I just put `1` as a case expression.

dalance, 2023-01-12:
I feel `case 1` may be bit strange, but I have no idea about alternative keyword.
So I'll keep the current implementation until new idea.

taichi-ishitani, 2023-01-12:
Introduce other keyword insead of `case`? For example, `switch`, `branch`..?

dalance, 2023-01-15:
`switch` seems to be good.

dalance, 2024-04-16:
The idea of `switch` statement moved to #643.