veryl-lang/veryl #12: remove () in always_ff

dalance, 2022-12-22:
Removing () in always_ff is uniform as `if` statement.

```
always_ff (clk) {
}
always_ff (clk, rst) {
}
always_ff (posedge clk, async_low rst) {
}
```

to

```
always_ff clk {
}
always_ff clk, rst {
}
always_ff posedge clk, async_low rst {
}
```

AlphaLyrae0, 2022-12-24:
What about removing { and } as well ? I'd prefer indent based block recognition like Python.

dalance, 2022-12-24:
> How about removing { and } as well ? I'd prefer indent based block recognition like Python.

Indent style like Python is very difficult to implement stable parser and formatter.
I think improving tool implementation smoothly is better than key typing reduction.

dalance, 2022-12-25:
If reset exists (especially polarity exists), removing () may be hard to read.

closed 2022-12-25 by dalance