veryl-lang/veryl #27: let statement

dalance, 2022-12-23:
let statement for variable declaration and module/interface instantiation.

```
// variable declaration
let value_a: logic [10];

// assignment after declaration
assgin value_a = 1;

// variable declaration with assignment
let value_b: logic [10] = 1;

// module instantiation
let module_a: inst moduleA;

// interface instantiation
let if_a: inst interfaceA;

// interface array
let if_b: inst interfaceB [10];

// module with parameter and port connection
let module_b: inst moduleB #(
    a,
    b: 10,
) {
    a,
    b: 10,
};

```

dalance, 2022-12-27:
Alternative idea: `new` operator in C++ style

```
let module_b = inst module B #(
..
};
```

Comparison

```
a: logic;
b: logic[10];
c: inst xx_if;
d: inst xx_if[10];

a: logic;
b: logic[10];
c = inst xx_if;
d = inst xx_if[10]; 
```

I think the original idea is better in interface instantiation at integrity point of view.

In C++ style, port connection is initialization value.
In the original idea, port connection is type variable.
I think type variable is better because the port connection is immutable.

dalance, 2022-12-27:
Reconsideration

`let` is variable declaration without port  connection. This can be used for bultin types, struct, enum, module, interface. 
`inst` is instantiation with port connection.

```
// variable declaration
let value_a: logic [10];

// assignment after declaration
assgin value_a = 1;

// variable declaration with assignment
let value_b: logic [10] = 1;

// module instantiation without port
let module_a: moduleA #(a: 1, b: 2);

// interface instantiation without  port
let if_a: interfaceA;

// interface array
let if_b: interfaceB [10];

// module with parameter and port connection
inst module_b: moduleB #(
    a,
    b: 10,
) {
    a,
    b: 10,
};
```

dalance, 2022-12-28:
> `let` is variable declaration without port connection. This can be used for bultin types, struct, enum, module, interface. `inst` is instantiation with port connection.

In the transpiled SystemVerilog, `let` with module / interface requires `()`.
So the unified style doesn't work fine.

```
let if_a: interfaceA;
let value_a: logic;

↓

interfaceA if_a();
logic value_a;
```

dalance, 2022-12-28:
`inst` is for module / interface, and `let` for others.

```
// variable declaration
let value_a: logic [10];

// assignment after declaration
assgin value_a = 1;

// variable declaration with assignment
let value_b: logic [10] = 1;

// module instantiation without port
inst module_a: moduleA #(a: 1, b: 2);

// interface instantiation without  port
inst if_a: interfaceA;

// interface array
inst if_b: interfaceB [10];

// module with parameter and port connection
inst module_b: moduleB #(
    a,
    b: 10,
) {
    a,
    b: 10,
};
```

taichi-ishitani, 2023-01-05:
SystemVerilog has a similar keyword `var`. How about using `var` instead of `let`?

taichi-ishitani, 2023-01-05:
[image]

This is a code snippet from SystemVerilog LRM.
The `var` keyword is used for the same purpose of the `let` keyword.

dalance, 2023-01-05:
It seems to be good.

```
// variable declaration
var value_a: logic [10];

// assignment after declaration
assgin value_a = 1;

// variable declaration with assignment
var value_b: logic [10] = 1;

// module instantiation without port
inst module_a: moduleA #(a: 1, b: 2);

// interface instantiation without  port
inst if_a: interfaceA;

// interface array
inst if_b: interfaceB [10];

// module with parameter and port connection
inst module_b: moduleB #(
    a,
    b: 10,
) {
    a,
    b: 10,
};
```

dalance, 2023-01-08:
Port connection of `inst` should be `()` instead of `{}` because module port declaration uses `()`?

```
// module with parameter and port connection
inst module_b: moduleB #(
    a,
    b: 10,
) (
    a,
    b: 10,
);
```

closed 2023-01-10 by dalance