veryl-lang/veryl #28: Not report a syntax error when parsing VL code including syntax error

taichi-ishitani, 2022-12-24:
I got a problem that `veryl build` command does not report a error when parsing VL code including a syntax error.

How to reproduce:

Run `veryl build` command with following VL code.

```
module test_module (
 i_clk: input logic,
 i_rst_n: input logic,
 i_d: input logic,
 o_d: output logic
) {
  d: logic;

  always_comb {
    o_d = d;
  }

  always_ff (i_clk, i_rst_n) {
    if_rest {
      d = 0;
    }
    else {
      d = i_d;
    }
  }
}
```

I expect the command reports a syntax error because there is a typo (if_reset -> if_rest).
But the command completes SV code generation.

dalance, 2022-12-24:
keyword detection in lexer was wrong. I fixed it.

taichi-ishitani, 2022-12-25:
Thank you. I will check later.

dalance, 2022-12-25:
I added semantic check before build.

taichi-ishitani, 2022-12-25:
I confirmed this issue is fixed by the latest commit da50b37ba55b22905b40b356ae6382dca3352b53.
Thank you for your fixing.

closed 2022-12-25 by taichi-ishitani