veryl-lang/veryl #29: Using SV sytem functions

taichi-ishitani, 2022-12-24:
SystemVerilog has useful sytem functions, such as `$clog2`.
I'd like to use SV system functions within VL code but I got a syntax error.

```
Error: parol_runtime::unexpected_token

  × Unexpected token: LA(1) (Error)
   ╭─[test.vl:3:1]
 3 │   parameter MAX_COUNT:   u32 = 8,
 4 │   parameter COUNT_WIDTH: u32 = $clog2(MAX_COUNT + 1)
   ·                                ┬
   ·                                ╰── Unexpected token
 5 │ )(
   ╰────
  help: Unexpected token
```

Does not support such feature?

dalance, 2022-12-25:
General function call is not still supported.
System function call will be supported at the same time.

taichi-ishitani, 2022-12-26:
OK, I understand.
I track this request by using #33.

closed 2022-12-26 by taichi-ishitani