veryl-lang/veryl #30: Part select using position and width

taichi-ishitani, 2022-12-24:
Part select using position and width (`a[pos+:WIDTH]`/`a[pos-:WIDTH]`) is useful feature.
So I'd like you to import this feature to veryl.

In addition, I think part select using index and width like below is a code showing often.

```systemverilog
b = a[2*i+:2];
```

Therefore, I'd like you to introduce a syntax sugger for this code like below.

```
b = a[2, i]; // a[2*i+:2]
```

dalance, 2022-12-26:
`[2, i]` or `[i, 2]` may be confusable.
It means "index i @ step 2", so how about `[i step 2]` or `[i width 2]`

taichi-ishitani, 2022-12-26:
`step` has already been used as a keyword so I think `[i step 2]` is better notation.

dalance, 2022-12-27:
Added `+:` `-:` and `step`

closed 2022-12-29 by dalance