veryl-lang/veryl #37: Generated SV code with comments is illegal

taichi-ishitani, 2022-12-26:
The build commmand genertes illegal SV code from VL code with comments.
These comments are put:

* between else and if of else-if statement
* before condition of if statement

These are an exaple code and generated SV code:

```
module test_module (
 i_clk: 	input logic,
 i_rst_n:	input logic,
 i_up:	input logic,
 i_down:    input logic,
 o_count:	output logic[8]
) {
  count: 	logic[8];
  up_down: 	logic[2];

  always_comb {
    up_down = (i_up << 1) | i_down;
  }

  always_ff (i_clk, i_rst_n) {
    if_reset {
      count = 0;
    }
    else // count up
    if up_down == 2'b10 {
      count = count + 1;
    }
    else if // count down
    up_down == 2'b01 {
      count = count - 1;
    }
  }
}
```

```systemverilog
module test_module (
    input  logic         i_clk  ,
    input  logic         i_rst_n,
    input  logic         i_up   ,
    input  logic         i_down ,
    output logic [8-1:0] o_count
) ;
    logic [8-1:0] count  ;
    logic [2-1:0] up_down;
    always_comb begin
        up_down = (i_up << 1) | i_down;
    end
    always_ff @ (posedge i_clk, negedge i_rst_n) begin
        if (!i_rst_n) begin
            count <= 0;
        end else // count up if (up_down == 2'b10) begin
            count <= count + 1;
        end else if // count down (up_down == 2'b01) begin
            count <= count - 1;
        end
    end
endmodule
```

`if (up_down == 2'b10) begin` and `(up_down == 2'b01) begin` are hidden because they are comments.
I think `/* */` comment format should be used for inserted comments.

dalance, 2022-12-26:
The root cause is formatting algorithm.
`veryl fmt` will be the simillar illegal result.
Formatting line comments inserted to arbitrary position is difficult...

dalance, 2022-12-26:
This was fixed.

taichi-ishitani, 2022-12-26:
Thank you for the fix.
I confirmed that the generated SV is legal code.

closed 2022-12-26 by taichi-ishitani