veryl-lang/veryl #45: Need to insert type cast

taichi-ishitani, 2022-12-27:
Inserting type cast is needed when LHS of assignment is enum variable and RHS of assignment is non-enum value.
If no type cast is inserted then EDA tool may report a warning like below.

```
Warning-[ENUMASSIGN] Illegal assignment to enum variable
testbench.sv, 13
test, "foo_bar = 0;"
  Only expressions of the enum type can be assigned to an enum variable. 
  The type int is incompatible with the enum 'e_foo_bar'
  Expression: 0
  Use the static cast operator to convert the expression to enum type.
```

However generated SV code does not include type cast.

orignal VL code:

```
module test {
  enum e_foo_bar: logic [1] {
    foo = 0,
    bar = 1
  }

  let foo_bar: e_foo_bar;
  always_comb {
    foo_bar = foo;
    foo_bar = bar;
    foo_bar = 0;
    foo_bar = 1;
  }
}
```

generated SV code:

```systemverilog
module test ;
    typedef enum logic [1-1:0] {
        foo = 0,
        bar = 1
    } e_foo_bar;
    e_foo_bar foo_bar ;
    always_comb begin
        foo_bar = foo;
        foo_bar = bar;
        foo_bar = 0;
        foo_bar = 1;
    end
endmodule
```

dalance, 2022-12-27:
I think explicit type cast is better than implicit.
So `foo_bar = 0`  will be semantic error.

Explicit type cast is not implemented yet.

dalance, 2022-12-27:
#48

taichi-ishitani, 2022-12-27:
OK, I understood.

closed 2022-12-27 by taichi-ishitani