veryl-lang/veryl #46: enum value range check

taichi-ishitani, 2022-12-27:
The `veryl build` command generates SV code even if an enum value is outrange.

orignal VL code:

```
module test ;
    typedef enum logic [1-1:0] {
        foo = 0,
        bar = 2
    } e_foo_bar;
endmodule
```

generated SV code:

```systemverilog
module test ;
    typedef enum logic [1-1:0] {
        foo = 0,
        bar = 2
    } e_foo_bar;
endmodule
```

The verly command needs to check range of enum values because this causes a compile error.

```
Error-[ENUMRANGE] Enum label outside value range
testbench.sv, 7
  The enum label 'BAR' has the value 2 which is outside the range of the base 
  type of the enum 'e_foo_bar', which is 1 bit unsigned.
  If this is indeed the intended value, consider expressing it as as sized 
  literal of the same width and signedness as the enum base type.
```

dalance, 2022-12-27:
This check require #47, so will be implemented after it.

taichi-ishitani, 2022-12-27:
I understood.
I will trace this issue by using #47.

dalance, 2022-12-27:
I reopen this issue to remind the check implementation.

closed 2023-01-18 by dalance