veryl-lang/veryl #59: import declaration

dalance, 2023-01-06:
```
import PackageA::Item;
import PackageB::*;
```

module's import like below causes parsing and formatting complexity.

```
module ModuleA import PackageA::* (
) {
}
```

Instead of supporting module's import, is it sufficient like below? 

```
import PackageA::*;

module ModuleA (
) {
}
```

taichi-ishitani, 2023-01-12:
Visibility of items imported by import declarations outside module declarations is global.
In fact, compiling following SV files does not report a compile error.

```systemverilog
// foo_pkg.sv
package foo_pkg;
  localparam FOO = 0;
endpackage
```

```systemverilog
// foo_0.sv
import foo_pkg::*;
module foo_0;
  localparam FOO_0 = FOO;
endmodule
```

```systemverilog
// foo_1.sv
module foo_1;
  localparam FOO_1 = FOO;
endmodule
```

Therefore, I think import declarations should be inside a module declaration.

dalance, 2023-01-12:
There are three ideas.

* Allow import at module declaration header like SV
* Implement file scope import

```
import PackageA::*;

module ModuleA (
) {
}

module ModuleB (
) {
}
```

to

```systemverilog
module ModuleA import PackageA::*;
endmodule

module ModuleB import PackageA::*;
endmodule
```

This case disallows global import like SV.

* Expand import scope in module to module port declaration

```
module ModuleA (
) {
    import PackageA::*;
}
```

to

```systemverilog
module ModuleA import PackageA::*;
endmodule
```

In this case all imports in a module move to the module header.

taichi-ishitani, 2023-01-12:
I think `Implement file scope import` is a better choise.

> This case disallows global import like SV.

I think this is not a problem because large scope has bad influence.

closed 2023-01-17 by dalance