veryl-lang/veryl #62: Useage of tri keyword

taichi-ishitani, 2023-01-07:
I think in Varyl `tri` keyword is used as a data type.
On the other hand, in SystemVerilog the `tri` keyword is used to show that a declared element is a wire and to allow the element to have multiple drivers. Therefore, its usage is similar to the usage of the `var` keyword.

```systemverilog
package foo_pkg;
  typedef enum logic {
    FOO_0,
    FOO_1
  } e_foo;
endpackage

module foo_sub
  import foo_pkg::*;
(
  inout tri e_foo io_foo
);
  assign io_foo = FOO_0;
endmodule

module foo_top;
  import foo_pkg::*;

  tri e_foo foo;
  assign foo = FOO_1;
  foo_sub u_sub (foo);
endmodule
```

I think Veryl should also use the `tri` like the `var` keyword.

dalance, 2023-01-07:
`tri` may be type modifier like `const` in C++.

```
module A (
    a: inout tri logic[10],
    b: inout tri struct_t,
) {
    var a: tri logic;
    var b: tri struct_t;
}
```

Usage `tri` like `var` keyword is not suitable because `var` can't be used in port declaration.

Syntactically `tri u32` is allowed, but it will be rejected by semantic checker.

dalance, 2023-01-07:
Alternatively, `inout` infers `tri` automatically.
This may be less consistency?

```
module A (
    a: inout logic[10],
    b: inout struct_t,
) {
    tri c: logic;
    tri d: struct_t;
}
```

dalance, 2023-01-07:
I think `tri` is special case, and is required to write carefully.
So explicit description like the first idea is better?

taichi-ishitani, 2023-01-08:
> var d: tri struct_t;

I feel strange that var and tri are used in the same object declaration because var means a variable.
I also think `var` should be a type modifier if `tri` is a type modifier.

According to the above thought, how about this?

* change `var` to a type modifier
    * this is for symmetry with `tri` keyword
* revive `let` keyword

```
module A (
  a: output var logic, // var can be omitted
  b: output tri logic
){
  let c: var logic; // var can be omitted
  let d: tri lgoic;
}
```

dalance, 2023-01-08:
I think "omittable" is a main cause of SystemVerilog's syntax complexity.
So `var` should be mandatory or like below

```
module A (
  a: output logic,
  b: output tri logic
){
  let c: logic;
  let d: tri lgoic;
}
```

I think symmetricity is not required because `tri` is special case.
(for example, many languages mark `static` variable but not `dynamic` variable)

> I feel strange that var and tri are used in the same object declaration because var means a variable.

I read `var d: tri struct_t;` as "variable declaration of tri-statable struct_t".
So `var` is keyword to show variable declaration, and `tri` is type modifier.

 I agree the meaning of `var` of SV and Veryl is a little different.
If the difference causes strangeness, I agree to revive `let`.

taichi-ishitani, 2023-01-08:
> If the difference causes strangeness, I agree to revive let.

I think there are few SV users using `var` keyword so this could not be a problem.
Therefore, it's OK for me whichever using `var` or `let` keywords.

dalance, 2023-01-09:
Comparison of `var` and `let`

```
var  a: u32;
inst b: InterfaceA;

let  a: u32;
inst b: InterfaceA;
```

I think `var`/`inst` is more clear that it is "variable".
(If `let` is used, I want to use `let` for instantiation too. but it can't)

taichi-ishitani, 2023-01-10:
> I think var/inst is more clear that it is "variable".

Yes, I agree.

dalance, 2023-01-11:
tri as type modifier was implemented.

closed 2023-01-11 by dalance