veryl-lang/veryl #69: zero width variable

dalance, 2023-01-11:
```
module ModuleA #(
    parameter W: u32 = 0,
) {
    var a: logic [W];
}
```

In SystemVerilog, this code will be error.
The transpiled representation should be considered.

taichi-ishitani, 2023-01-12:
`logic [-1:0] a` is a valid SystemVerilog code but it is very strange code.
I think `W` should be a positive value and Veyl should report a warning or an error.

dalance, 2023-01-12:
Reporting warning or error is difficult because Veryl can't know the actual parameter value including override.
For example, suppressing zero width range select by `[(W==0)?0:W-1:0]` instead of `[W-1:0]` may be available.

taichi-ishitani, 2023-01-12:
`logic [-1:0] a` does not cause a compile error so I think `[(W==0)?0:W-1:0]` is unnecessary.
Insread of `[(W==0)?0:W-1:0]`, how about inserting check logic to the generated SV code like below.

```systemverilog
module ModuleA #(
  parameter int W = 0
);
  initial assume(W > 0);
endmodule
```

It's OK for synthesis because synthesis tools may skip this kind of code.

dalance, 2023-01-12:
`logic [-1:0] a` causes elaboration error in some synthesis tools.
So I wrote `[(W==0)?0:W-1:0]` by hand to avoid it.

dalance, 2023-01-13:
move to #73

closed 2023-01-13 by dalance