veryl-lang/veryl #73: [要望] 幅ゼロの指定

ryuz, 2023-01-12:
C++ のサイズ0の配列のように、幅ゼロのバスを定義できると嬉しいです。
SystemVerilogの場合 たとえば 1対N マルチプレクサ を

```
module multiplexer
    #(
        parameter   int N = 2,
        localparam  int S = $clog2(N)
    )
    (
        input   logic   [N-1:0] in,
        output  logic   [0:0]   out,
        input   logic   [S-1:0] sel
    );

    assign out = in[sel];

endmodule
```

と書きたいところですが、 N が 1 の時に S が 0 になってしまい合成エラーになるため

```
module multiplexer
    #(
        parameter   int N = 1,
        localparam  int S = N > 1 ? $clog2(N) : 1
    )
    (
        input   logic   [N-1:0] in,
        output  logic           out,
        input   logic   [S-1:0] sel
    );

    generate
    if ( N > 0 ) begin : select
        assign out = in[sel];
    end
    else begin : bypass
        assign out = in;
    end
    endgenerate
    
endmodule
```

などと冗長な書き方をすることがあります。

また、変数を結合するのに例えば、下記のようなケースで b が不要な時に 0 を指定するとエラーになります。

```
module concatenate3
    #(
        parameter   int A = 2,
        parameter   int B = 0,
        parameter   int C = 5
    )
    (
        input   logic   [A-1:0]     a,
        input   logic   [B-1:0]     b,
        input   logic   [C-1:0]     c,
        output  logic   [A+B+C-1:0] q
    );

    assign q = {a, b, c};

endmodule
```

これらは、Verilog が [0:0] なバス幅指定をすることに起因していますが、Verly は Rust 風の宣言なので、解決しないものかと考えた次第です。
ただし、あくまでトランスパイらなので、変換を考えると難しいというのも想像できます。

例えばですが parameter を、そのまま SV に残すものと、Verly  内で解釈するものとを指定できるようにして、後者に限定して高度なことができるような事などではできないものでしょうか？

優先度の高い要望ではないのですが、何かの折にもし良いアイデアが出てくればご検討いただければ幸いです。

taichi-ishitani, 2023-01-12:
以下の様に生成するようにできれば、解決できるとは思いますが、生成後のコードが読みづらくなるのが難点ではあります。

```systemverilog
out = in[(S>0)?sel:0];
```

```systemverilog
q = {{(A>0){a}}, {(B>0){b}, {(C>0){c}}};
```

ryuz, 2023-01-12:
ありがとうございます。こういうテクニックは思いつかなかったので非常にありがたいです。

もっとも

```
input   logic   [B-1:0]     b,
```

部分が

```
input   logic   [0-1:0]     b,
```

と展開されてしまい、本来の趣旨と違う信号が生成されてしまうという問題は残りますが。

> 生成後のコードが読みづらくなるのが難点ではあります。

結局、ここに尽きるのかなと思いました。
 Veryl の使用用途も

- そのまま使える奇麗な SV に変換したい時
- 波形ファイルさえ Veryl と対応していればOKな時

の2種類ありそうな気がしますので、後者に踏み込んだオプションがあっても嬉しいです。

ちなみに Rust 風のマクロを実装したりなどもお考えでしょうか？
結構、この手のテクニックの隠ぺいにマクロは便利そうな気もしております。

dalance, 2023-01-12:
幅0のconcatenateは面白いですね。
全部これに対応すると確かに可読性が厳しくなってしまうので、 `zeroable` みたいなtype modifierを付けた変数だけ展開するとかをイメージしていました。

マクロについてはRustの手続きマクロのようにアトリビュートを付けた構文要素の構文木変換として実装したら良さそうかと思っています。

taichi-ishitani, 2023-01-13:
`zeroable` 良いですね。
これが指定されている場合は、

* 幅が 0 の場合に
    * 幅を強制的に 1 にするコードを挿入する
        * `logic [((W>0)?W-1:0):0] a`
    * 左辺値として参照する際に、強制的に 0 にするコードを挿入する
        * `lhs = (W > 0) ? a : '0`
    * 右辺として代入される際は、代入を無かったことにするコードを挿入する
        * `a = (W > 0) ? lhs : '0`
    * 連接の場合は、回数 0 の繰り返しを挿入して、辻褄を合わせる
        * `{{(A>0){a}}, {(B>0){b}}, {(C>0){c}}}`
        * 回数 0 の繰り返しが使える条件は限られているようなので、連列自体を禁止にしてしまった方が良いかもしれない

が良いでしょうか？ 

`zeroable` が指定されていない変数については、幅が 1 以上前提ですから、ポカミス除けに、

```systemverilog
initial assume(W > 0);
```

のようなコードを挿入するはどうでしょうか？

dalance, 2023-01-13:
良さそうですね。最初のコード例では

```
module multiplexer #(
    parameter  N: u32  = 2        ,
    localparam S: u32  = $clog2(N),
) (
    i: input  logic [N],
    o: output logic [1],
    s: input  zeroable logic [S],
) {

    assign o = i[sel];

}
```

```systemverilog
module multiplexer #(
    parameter  int N = 2,
    localparam int S = $clog2(N)
) (
    input  logic [N-1:0] i,
    output logic [1-1:0] o,
    input  logic [((S>0)?S-1:0):0] s
);

    assign o = i[(W>0)?s:'0];

endmodule
```

なので可読性もそこまで悪くはないかな、と思います。
マクロはマクロでいろいろ問題がありますし。
`$clog2` みたいな感じの関数があれば良かったんですけどね。

`initial assume` もいいと思います。

taichi-ishitani, 2023-01-13:
> $clog2 みたいな感じの関数があれば良かったんですけどね。

生成する SV で使うユーティリティ関数をまとめた pakcage を用意するのはどうでしょうか？
しかし、参照する際と、代入する際の処理はマクロにせざるを得ないなので、御利益はあまりなさそうですね。

ryuz, 2023-01-13:
いろいろとご検討いただきありがとうございます。zeroable のアイデアは素敵に思います。

私が FPGA ターゲットなのもあるかもしれませんが、汎用性の高いモジュールを書きたい要望が高いです。
普段  parameter や generate を多用することが多いのですが

- $clog2 が 1 に対して 0 になってしまう問題
- AXIバスのようにオプション信号が多い規格で使わない信号を幅0定義したら無きものとして扱いたい
- 機能追加するときは後方互換性を確保したいのでデフォルト引数が欲しい

などの需要があります。

最後の奴は SV は function や task ではデフォルト引数指定できるのに module のポートではできないようなので、parameter の方で指定しない限りは信号は無視するようなやり方で拡張したりしています。
(デフォルト引数の処理は、module 側ではなく、インスタンシエートする側の処理なので Veryl では拡張は難しそうですよね。)

ちなみにサイズ0に関しては $bits の問題もあって、ゼロの時に $bits が 0 を返すようにするのは少し難しい気がします。

そもそもの話として SV で幅 W の信号を [W-1:0] と表したときに

```
module tb_top();
    generate
    for ( genvar i = -2; i <= +2; ++i ) begin : foos
        foo #(.W(i)) i_foo ( .a('0) );
    end
    endgenerate
    initial #1 $finish();
endmodule

module foo #(W=1) (input [W-1:0] a);
    initial $display("W=%d $bits(a)=%d", W, $bits(a));
endmodule
```

を動かすと

```
W=         -2 $bits(a)=          4
W=         -1 $bits(a)=          3
W=          0 $bits(a)=          2
W=          1 $bits(a)=          1
W=          2 $bits(a)=          2
```

のように 0以下でおかしなことになるのですよね。
(SV の文法が示しているのはbitサイズではなくbit位置だというのがそもそもの問題で

多くの場合、負の値を使うこと自体サポート範囲外にしてよい気がするので

> initial assume(W > 0);

はとても良い案に思いました。

taichi-ishitani, 2023-01-13:
本筋と外れますが、

> AXIバスのようにオプション信号が多い規格で使わない信号を幅0定義したら無きものとして扱いたい

社内用のバスインターフェース用の共通モジュール作成の時にも苦慮した点ですね。
インターフェース上の信号は消せないのは仕方ないとして、FIFO などに繋ぐ際に、必要ビットだけビット列に詰めることで対応しました。これによって、必要最小限のリソースだけを消費するようにしています。
詰める function、元に戻す funtion を SV interface に実装することで、関数を呼び出すだけで、これらを実施できるようにしています。

Qiita の記事や、実際の実装が参考になるかと思います。

* https://qiita.com/taichi-ishitani/items/ec351cc5d793a4382da0#%E6%8E%A5%E7%B6%9A%E3%81%AE%E7%B0%A1%E7%95%A5%E5%8C%96-%E3%83%93%E3%83%83%E3%83%88%E5%88%97%E3%81%AB%E8%A9%B0%E3%82%81%E3%82%8B%E7%B7%A8
* https://github.com/pezy-computing/pzbcm/blob/master/pzcorebus_common/pzcorebus_if.sv
* https://github.com/pezy-computing/pzbcm/blob/master/pzcorebus_common/pzcorebus_if_internal_macros.svh
* https://github.com/pezy-computing/pzbcm/blob/master/pzcorebus_fifo/pzcorebus_fifo.sv

ryuz, 2023-01-13:
> 社内用のバスインターフェース用の共通モジュール作成の時にも苦慮した点ですね。

やはり似たような課題は皆さんお持ちなのですね。
SV対応してないツールの対応もあって、Verilog で パック/アンパックするコンポーネント書いて、AXI-Stream 信号を FIFO に投げ込むとかやってました。

https://github.com/ryuz/jelly/blob/master/rtl/library/jelly_func_pack.v
https://github.com/ryuz/jelly/blob/master/rtl/library/jelly_func_unpack.v
https://github.com/ryuz/jelly/blob/master/rtl/bus/jelly_axi4s_fifo.v

この手の課題はコードが汚くなりがちなので、うまいことバグなく綺麗に書ける仕組みがあると嬉しいですね。

taichi-ishitani, 2024-03-08:
#528 と同じ用途なので、こちらにもコメントを。
`zeroable` の様に型修飾子を付ける場合、関連する全ての変数に修飾子を付ける必要があると思います。
付けないと、以下のように、幅不一致が起きてしまう恐れがあります。

```systemverilog
// W=1 の時、count_next は logic [-1:0] になってしまう
var count: zeroable logic<W>;
var count_next: logic<W>;
always_comb {
  count_next = count + 1;
}
```

なので、付け忘れを検知する機能や、#528 で提案されている `(W > 2) ? $clog2(W) : 1` を返す組み込み関数がいるのではないかと思います。