veryl-lang/veryl #85: signed / unsigned representation

dalance, 2023-01-16:
Signed / unsigned specifier is not implemented.

### `signed` / `unsigned` like SV

```
var a: signed logic[10];
var b: unsigned logic[10];
var c: logic[10]; // signed or unsigned ?
```

Some problems.

* bit long
* word length is not uniformed
* omitted case may be confusable

### `ilogic` / `ulogic`

This is the same as `isize` / `usize` in Rust.

```
var a: ilogic[10]; // signed logic in SV
var b: ulogic[10]; // unsigned logic in SV
var c: ibit  [10]; // signed bit in SV
var d: ubit  [10]; // unsigned bit in SV
```

This seems to be better than `signed` / `unsigned`?

taichi-ishitani, 2023-01-17:
I think using a signed value is a special case than using a unsigned value.
Therefore, I prefer introducing the `signed` keyword as a type modifier like the `tri` keyword.

dalance, 2023-01-17:
OK. It seems to be good that unsigned by default and `signed` is added only.

closed 2023-01-17 by dalance