veryl-lang/veryl #109: [質問] parameter で型指定無しにしたり、文字列を使う場合について

ryuz, 2023-01-29:
いつもお世話になっております。 Veryl 0.3.0 を試させて頂いております。

現在、下記の SystemVerilog と等価なものを出力する Veryl のコードを書こうと試みております。

https://github.com/ryuz/jelly/blob/8701d4830836b33d2fe19de9a8e6e330d2045e15/rtl/library/jelly2_ram_dualport.sv

その際に parameter に文字列が使いたいのですが Veryl のサンプルで文字列型を使っているものを見つけきれずよくわかりませんでした。

また

- 一方で string を指定すると古い合成器がエラーになるので形指定無しで parameter に文字列を使う場合がある
- 引数を attribute に利用することもある

などがあり、ご意見いただけるとありがたいです。

ちなみに Xilinx の ハードマクロでもパラメータ引数に文字列を要求するものがよくあります。
参考までに Xilinx のライブラリガイドのURLを下記 に記載いたします。

https://www.xilinx.com/content/dam/xilinx/support/documents/sw_manuals_j/xilinx2019_2/ug974-vivado-ultrascale-libraries.pdf

いつも要望ばかりで恐縮ですがよろしくお願いいたします。

dalance, 2023-01-29:
文字列リテラルは後で追加するつもりで忘れていました…。
追加したので問題なければ明日v0.3.1をリリースします。

形無しについてはちょっと考える必要がありますね。
少なくともデフォルトでは明示的であるべきだと思っているので
何らかのオプションで調整する形でしょうか。

* Very.tomlの設定
プロジェクト全体でparameterの型を全て省略するのであれば、Veryl.tomlで設定できるようにするのがいいと思います。

* アトリビュート
特定のparameterだけに適用するのであればアトリビュートでの指定になると思います。

taichi-ishitani, 2023-01-29:
右辺値が文字列であっても、logicかbitで受ければ良いので、型付け必須でもいいのではないでしょうか？

taichi-ishitani, 2023-01-29:
それか、TB記述を考慮しなくても良いのであれば、「parameter宣言内のstring型は、型無parameter宣言に変換する」も手でしょうか？

ryuz, 2023-01-29:
> 追加したので問題なければ明日v0.3.1をリリースします。

いつも迅速なご対応、ありがとうございます。

> 右辺値が文字列であっても、logicかbitで受ければ良いので、型付け必須でもいいのではないでしょうか？

logic だと幅の事前指定が必要で、"FALSE" や "TRUE" のような長さの違う文字列を渡すと、ビット長不一致ワーニングで lint が通らないという理由で避けていたと思います。
Veryl の方での bit 幅チェックが十分であれば、シンセサイザやシミュレータの方のワーニングはOFFできるかもしれませんが、 Veryl でない資産とも接続すると思うので、parameter 指定などで、後からでないと幅チェックできない部分が出てきそうな気もしております。

> それか、TB記述を考慮しなくても良いのであれば、「parameter宣言内のstring型は、型無parameter宣言に変換する」も手でしょうか？

Veryl.toml で設定できれば、そういうオプションはありかと思います。string が使える処理系ではそのまま通して、使えない処理系を使う時だけ型なしが選べると嬉しいです。

現状、形無しで渡してますが、Verilator などの幅チェックが、"FALSE" や "TRUE" のような長さの違う文字列を比較すると怒るので、一度 bit 幅拡張してから比較するとかの苦肉の策をしています。
Veryl で綺麗な書き方ができると嬉しいです。

taichi-ishitani, 2023-01-29:
文字列を使うのはenumが無かった時代の苦肉の策ですし、こういうverilogとの橋渡しな箇所はSVに押し付けるのも一手なのかなとも思います

dalance, 2023-01-30:
Veryl.tomlで指定するとして、全部省略するかどうかのon/offだけでいいですか？
それともそれぞれの型ごとの指定が必要でしょうか？

```
[build]
implicit_parameter_type = true # 全ての型を省略

implicit_parameter_type = ["string", "u32"] # `string` と `u32` だけ省略
```

taichi-ishitani, 2023-01-30:
enum は enum であって欲しいので、型を省略可能にするのであれば、特定の型だけ省略可能するほうが良いですね。

dalance, 2023-01-30:
implicit_parameter_typesを追加しました。

```
[build]
implicit_parameter_types = ["string", "u32"]
```

dalance, 2023-01-30:
v0.3.1リリースしました。

ryuz, 2023-01-30:
ありがとうございます。v0.3.1 にて下記のコードを SV に変換して、Vivado 2022.2 で問題なく合成できることまで確認できました。

この手のプリミティブな機能も Veryl で書けるというのは非常にありがたく思います。

もう少しいろいろ試してみたいと思います。引き続きよろしくお願いいたします。

```
module block_ram
#(
    parameter  ADDR_WIDTH : u32 = 10,
    parameter  DATA_WIDTH : u32 = 18,
    parameter  RAM_TYPE   : string = "block",
    localparam MEM_SIZE   : u32 = 2 ** ADDR_WIDTH,
) (
    clk  : input  logic,
    en   : input  logic,
    we   : input  logic,
    addr : input  logic<ADDR_WIDTH>,
    din  : input  logic<DATA_WIDTH>,
    dout : output logic<DATA_WIDTH>,
) {

    #[sv("ram_style=RAM_TYPE")]
    var mem : logic<DATA_WIDTH> [MEM_SIZE];

    always_ff (clk) {
        if en {
            if we {
                mem[addr] = din;
            }
            else {
                dout = mem[addr];
            }
        }
    }
}
```

closed 2023-01-30 by dalance