veryl-lang/veryl #110: [Feature] boolean type

taichi-ishitani, 2023-01-30:
The `boolean` data type is useful to declare variables/parameters which mean on/off.
Therefore, I'd like to introduce the `boolean` data type.

This is equivalent SV definition.

```systemverilog
typedef enum logic {
  VERYL_FALSE = 1'b0,
  VERYL_TRUE  = 1'b1
} veryl_boolean;
```

dalance, 2023-01-30:
Looks good.
Which is better `enum logic` and simple `logic`?

`enum logic` can check assignment without cast in SystemVerilog side.
But  I think the result of compare operator can't be assigned directly.
(Maybe can do it?)

```
var a: bool = (x > 1) as bool;
```

If Veryl insert the cast automatically, the transpiled code may be complicated.

```systemverilog
veryl_boolean a;
assign a = veryl_boolean'(x > 1);
```

`logic` can do it, and type check can be handled at Veryl side.
`true` and `false` will be converted to `0` and `1`. It has less explicity than `enum logic`.

taichi-ishitani, 2023-01-30:
Visivility of enum values on dump files is better than visivility of logic values so I prefer `enum logic` than `logic` a little bit.
Therefore, I don't care if you choose `logic`.

For readability, how about using following definition?

```systemverilog
tyepdef logic veryl_boolean;
localparam veryl_boolean VERYL_TRUE  = 1'b1;
localparam veryl_boolean VERYL_FALSE = 1'b0;
```

dalance, 2023-02-06:
```systemverilog
parameter logic ENABLE_FUNC_A = 1;
parameter veryl_boolean ENABLE_FUNC_A = VERYL_TRUE;
```

For me, the first example is explicit sufficientlly, and the second is bit redundant.
(Addisionally, identification between `VERYL_TRUE` and `VERYL_FALSE` is not easy because they are upper case.)

I think HDL doesn't have sufficient motivation to treat boolean as special type because part select generates boolean casually.

taichi-ishitani, 2025-03-12:
I'm working on it.
`bool`, `true` and `false` keywords will eimitted into `logic`, `1'b1 and `1'b0`.

dalance, 2025-03-14:
I agree to introduce `true` and `false` because they show it is boolean value more clearly than `1'b0` and `1'b1`.
About `bool` type, I have some concerns.

`bool` type is almost type alias of `logic`, but casting behaviour is only different. This small different may be confusable.
And the behaviour of `bool` packed array is not obvious.

```systemverilog
let a: logic<4> = 4'b1110;
let b: logic    = a as logic   ; // b == 0
let c: bool     = a as bool    ; // c == 1
let d: logic<2> = a as logic<2>; // d == 2
let e: bool <2> = a as bool <2>; // e == ?
```

taichi-ishitani, 2025-03-14:
I introduce bool type as a fixed type so bool type cannot be used for packed array.

dalance, 2025-03-14:
OK. I understood that `bool` is a type alias of `logic<1>` to express boolean value. I'll merge #1389.

closed 2025-03-14 by dalance