veryl-lang/veryl #112: [Feature] Bundle

philipaxer, 2023-01-31:
Hi,

several meta HDL languages have a Bundle type (see SpinalHDL, Chisel for example). This simplifies IO declaration. This is a suggestion to support a Bundle.

regards
Philip

dalance, 2023-01-31:
Thank you for your suggestion.

In my understanding, Chisel's Bundle type is almost the same as `struct` which is suppotted by SystemVerilog.
Already `struct` can be used in Veryl too.

```
struct BundledType {
    member_a: logic<10>,
    member_b: logic<20>,
}

var x: BundledType;
assign x.member_a = 0;
assign x.member_b = 0;
```

philipaxer, 2023-01-31:
Hi, 
this is not entirely correct. The IO direction in (system) verilog is defined at the module's port declaration. So essentially as a user you are forces to declare two structs, one for the output, one for the input. I.e. in your example `x` is either in or out but never partially in/out.

Bundles try to fix this by annotating IO direction next to members. This IMO simplifies port declaration tremendously. For verilog serialization, i would recommend to flatten this into a plain list of signals. Otherwise you would need to create structs.

regards
Philip

philipaxer, 2023-01-31:
Link for reference:
https://spinalhdl.github.io/SpinalDoc-RTD/dev/SpinalHDL/Data%20types/bundle.html

dalance, 2023-01-31:
If you want to mix direction, you can use `interface`.

```
interface BundledType {
    var member_a: logic<10>,
    var member_b: logic<10>,

    modport master {
        member_a: input,
        member_b: output,
    }

    modport slave {
        member_a: output,
        member_b: input,
    }
}
```

Veryl's target is SystemVerilog. So the `interface` is directly transplied to SystemVerilog's `interface`.
No flatten is needed.

philipaxer, 2023-01-31:
Interfaces in systemverilog only work horizontally well. I.e. you can connect two instances on the same hierarchy level. If you want to pass interfaces vertically, it is a pain. Example in pseuo code:
````
interface myif  (
...
modport master ...
modport slave ...
)

module A () begin
 
myif mif;
inst B (mif.slave)

some_process begin
a FSM that stimulates mif.master
end
end
```` 

 The reason is that interfaces are paritially objects, partically datatypes and not really thought through IMO. 
I think they can only be used for BFMs, but that it.

dalance, 2023-01-31:
I can't understand your pain yet. In my company, `interface` is used through the entire of large scale ASIC designs.
Could you explain the pain point in the pseudo code?

philipaxer, 2023-02-01:
Last time I used interface in RTL design was roughly 1.5-2 years ago and I ran into tooling issues. I guess the LRM allows what I am asking for but i remember that synthesis tools do not support the full subset of interfaces. I tried quartus at the time, i didnt try to push it through design compiler.

dalance, 2023-02-01:
OK. I understand.
I heard SystemVerilog support of Quartus is insufficient. 
Other vendors like Synopsys, Cadence and AMD(Xilinx) support interface sufficientlly.

The top priority of this project is interoperability with SystemVerilog.
This is because I want to use this language with SystemVerilog codebase in my company.  
So Verilog  mode which expand interface to flatten wires can be considered, but the priority is low.

dalance, 2024-01-05:
Move to #145

closed 2024-01-05 by dalance