veryl-lang/veryl #117: [要望] 変数のリセット初期化漏れを警告してほしい

ryuz, 2023-02-01:
例えば下記のように 変数 b のリセットが漏れた場合、仮に b が初期化不要であっても reset 時に b の値をキープするための余計なロジックが生成されてしまいます。

```
always_ff (clk, reset) {
    if_reset {
        a = 0;
    }
    else {
        a = b + 2;
        b = a + 1;
    }
} 
```

Veryl では if_reset という構文が導入されているので、リセット漏れが検出できそうな気がしましたが如何でしょうか？

余談ですが個人的には初期化不要なものも

```
always_ff (clk, reset) {
    if_reset {
        a = 0;
        b = 'x;
    }
    else {
        a = b + 2;
        b = a + 1;
    }
} 
```

として、リセットで不定値にしたほうがシミュレーション時にバグだししやすいのでそういう書き方をよくします(これも賛否ありそうですが)。

taichi-ishitani, 2023-02-01:
後半の記述ですが、仮にリセット無しFFが合成できたとしても、RTLとネットリストで挙動が異なるので、さすがにこの記述は使えないですね

ryuz, 2023-02-01:
> 後半の記述ですが、仮にリセット無しFFが合成できたとしても、RTLと合成結果とでリセット時の挙動が異なるので、さすがにこの記述は使えないですね

ここはFPGA固有の事情はあるかと思います。現状リセット接続するFFを最小限にするのが推奨されているので、いつリセットを掛けたかで何が入ってるかわからない状態のFFだらけの状態から検証が必要となり、不定値いれたり乱数入れたりが必要になります。

今のところ手元のFPGAツール(Vivado)の場合、合成も問題なくやってくれるので、研究開発とか原理検証用の機器に使う分には手っ取り早く検証確認して動かしたいときには便利だったりもします。賛否あるのは否定しません。

ASIC開発とかに使えないというのはおっしゃる通りかと思います。

dalance, 2023-02-01:
リセット不要なら単純にリセットなしにした方がFPGAでもASICでも使え、初期値は不定になるのでシミュレーションでの確認もできる気がしますがいかがでしょうか？

```
always_ff (clk) {
    b = a + 1;
} 
```

それはともかく、リセット漏れの警告は可能と思います。
警告なのかエラー相当なのかは要検討でしょうか。

taichi-ishitani, 2023-02-01:
「リセットの有無でalways文を分ける」っていうのはその通りなのですが、リセット以外の構造が同じ場合は纏めて記述したい場合を何とかできないですかね？

dalance, 2023-02-01:
確かにそうですね。
その場合、Veryl側の記法と生成されるSVをどうするかが問題になるかと思います。

Veryl側は単にリセットで値を指定しないのが簡単ですが、これだと元々の要望にあったような漏れとの区別がつかないです。
なにか「意図的にリセットしない」ことを示したい感じはします。

```
always_ff (clk, reset) {
    if_reset {
        a = 0;
        // bに関する何らかの記述
    }
    else {
        a = b + 2;
        b = a + 1;
    }
} 
```

あと、SV側はVeryl側で1つだったalways_ffを2つに分割することになるので、
可読性と元コードとの対応関係がどうか、というところですね。

taichi-ishitani, 2023-02-01:
`not_reset b` みたいにアラタナキーワードを導入するか、`b=b` のように自己代入で代用するか、でしょうか？

ryuz, 2023-02-01:
ご検討ありがとうございます。

> b=b のように自己代入で代用するか

この場合、reset が来ているときに値を変化させないようにするロジックが生成されてしまうように思います。
リセット無しの always_ff に分けたのと同じ効果にしたければ if_reset {} の外に書かないと駄目じゃないかと思いました。

taichi-ishitani, 2023-02-01:
> b=b

この記述を「リセット無しとする」と言う意味です。

ryuz, 2023-02-01:
> この記述を「リセット無しとする」と言う意味です。

理解いたしました。大変失礼したしました。

ryuz, 2023-02-03:
> 後半の記述ですが、仮にリセット無しFFが合成できたとしても、RTLと合成結果とでリセット時の挙動が異なるので、さすがにこの記述は使えないですね

この点についてもう少し考えてみました。
LSI開発ではFPGAなどのラピッドプロトタイミング開発と異なり、RTLと生成されるネットリストが一致していないと後工程で困ることが多々あるのかと想像しております。

一方で、シミュレーションに際しては、タイミングバグで don't cate な値をラッチするようなミスを早期発見するのに意図的な不定値代入は役立つ場合もあるように思っています。

例えば Veryl でオプション指定しない限りは取り除かれる不定値挿入指示の構文があれば検証時に役に立つケースがないだろうかと考えてみたのですが如何でしょうか？(オプションON/OFFの両方で検証を流して確認するなど)

別の視点で、FPGAプロトなどで生産性だけを考える場合、値が don't cate になるケースで何を入れておくのが論理圧縮に有効かを合成器に任せられるので不定値を使う合理性もあるのですが、生産性と品質にトレードオフがあるのでケースバイケースかとは思っております。

taichi-ishitani, 2023-02-03:
> オプション指定しない限りは取り除かれる不定値挿入指示の構文

x を入れる手段はあるわけですし、特別な指示子は不要ではないでしょうか？
`a = 'x` のような、意図的な不定の代入を一律で消すオプションがあれば十分ではないでしょうか。

taichi-ishitani, 2023-02-03:
> オプション指定しない限りは取り除かれる不定値挿入指示の構文

ご指摘の通り、ASIC 開発においては、RTL とネットリストの一致は重要な項目です。
なので、生成後の SV に意図的な x の代入があると、流用するのが難しくなります。
どちらかと言うと、意図的な x の代入を一律で消してくれる方が有難いです。

`a = 'x` の様に x の代入手段はありますし、わざわざ別の指示子を導入するよりかは、x の意図的な代入を一律で消すオプションで十分ではないでしょうか？

ryuz, 2023-02-04:
コメントありがとうございます。

> a = 'x の様に x の代入手段はありますし、わざわざ別の指示子を導入するよりかは、意図的な x の代入を一律で消すオプションで十分ではないでしょうか？

おっしゃる通りかと思いました。ただ上の方で「リセット無し」記述をご検討されていたので、if_reset の中での x 代入に限り、「削除ではなくリセット無しとして扱う」のような手も含めて纏められないものかと少し思った次第でした。

x 代入も 波形ビューワでの目視はわかりやすいのですが、if 判定などでは単に偽になるだけなので偶然テストを抜けることもあり、Verilator など乱数を入れてくれるものも併用してテストしたりしています。これも完全ではないのでRTL記述段階からコントロールが効けば便利なケースも無いかと思ってみたのですが、ifdef 的なものを使うことも含めて代替手段はいろいろあるかと思っております(ですので、「何かのついででなら検討しても面白いかも」程度に捉えて頂ければ有難いです)。

もしトランスパイル時にいろいろやらせるのであれば

- Xを単純削除
- Xのまま残す
- X削除時に入る値の反転値を入れるコードに置き換える
- $random などの乱数を入れるコードに置き換える

などのバリエーションがあれば、一部はシミュレータだけではできないテストが出来そうに思いました。
(当方が、FPGAベンダーツールと 無償のOSS しか扱ったことがなく、ASIC用のEDAツールの世界だともっと違う解決方法があるのかもしれませんが)

taichi-ishitani, 2023-02-04:
> 「削除ではなくリセット無しとして扱う」

* リセットが不要
* 意図的に x を入れている

との区別がつかないですから、やはり記述上の区別は必要ではないでしょうか？
個人的な感覚ですが、「x の代入」はイレギュラーな記述なので、「リセットが不要」と「x の代入」の記述上の区別は欲しいところです。

taichi-ishitani, 2023-02-04:
もう一点。

```
if_reset {
  a = 'x
}
```

ですと、字面上は「２回目以降のリセットでも x になる」となります。
しかし、これを「リセットなしと扱う」とすると、実際の動作は「リセットが入力されても前の値を保持」となります。
なので、字面上の動作と実際の動作がことなるので、「if_reset 中の x 入力をリセットなし」とするのは不適当ではないでしょうか。

ryuz, 2023-02-04:
> なので、字面上の動作と実際の動作が異なるので、「if_reset 中の x 入力はリセットなし」とするのは不適当ではないでしょうか。

苦し紛れのいいわけをさせて頂くと、コンパイラは X の値をどう扱っても合法ですので、リセットなしに翻訳しても合法ではあると言えなくはないかとも思いました。
これは Veryl をコンパイラの一部ととらえるか、あくまでトランスパイラ (有効なSVを得たい)と捉えるかで変わってくるような気がしており、個人的にはトランスパイラとして使いたいのでおっしゃるご指摘その通りかと思います(やめた方がよさそうですね)。

taichi-ishitani, 2023-02-04:
デバッグの際はSVをみるわけで、Veryl上で見える動作と、SV上の動作が違うのは、混乱の元になり得ます。

また、「xの意図的な代入」と、「リセットなし」は、意図が異なるので、代用させないほうが良いと考えます。
Verilogのalways文にFF記述と組合せ回路記述の両方を担わせた例を考えれば、理解いただけるかと思います。

ryuz, 2023-02-04:
> デバッグの際はSVを見るわけで、

なるほど確かにその通りですね。

上で出ていた

> b=b
> この記述を「リセット無しとする」と言う意味です。

も同じことかと思いました(こちらはあくまで、たとえ話かとは思いますが)。
リセットの方も何かスマートな落としどころに落ち着くと嬉しく思います(妙案思いつかず申し訳ないのですが)。

taichi-ishitani, 2023-02-04:
「用途とは違う機能を担わせる」と言う観点では、よろしくない提案ですね。

- 次点の案である
- 「リセットが入っても前の値を保持」という挙動は守られている

という点を踏まえて提案した次第です。

dalance, 2023-02-06:
いろいろと難しいですね。
まずリセットなし宣言の構文についてですが、

* `not_reset b` のような特別な構文
  例によって代入文は構文的には結構深い位置にある（つまりalways_ffに限らずあらゆる位置に出現可能）ので
  特別な構文の導入コストは比較的高いです。
  `b = not_reset;` のようの代入文的にする方が多少ましですが、これもこれで任意の式中に `not_reset` を許容することになります。

* `b=b`
  構文的にはこちらの方がいいですが、通常の文が特別な意味を持つというのがどうか、という点はあります。
  例えば `b=(b);` や `b={b};` 、 `b=b+0;` はどう判断すべきか、といった部分です。

* `#[not_reset]` アトリビュート
  構文への影響を少なく指定する方法としてアトリビュートは考えられますが、アトリビュートの付与先がありません。

また生成されるSV側の問題として

* 論理の重複
  Veryl側で1つの `always_ff` が2つに分割されるので、内部の論理はそのままコピーされます。
  Veryl側で2つに分けていれば、その時点の可読性に応じてfunction化したり一旦 `always_comb` で複雑な論理を切り出したりするかもしれませんが、1つなら許容できるかもしれません。
  つまり可読性に関する判断がVerylとSVでずれてしまう気がします。

* コメントの扱い
  現在Veryl側のコメントはSV側のなるべく適切な位置に出力するようにしていますが、あまり大規模なコード改変が起きるとこれは難しくなります。
  コメントはコメント対象に対して前置と後置両方のケースがあるので、あるコメントがどの文に付いているのかを正確に判断するのは困難です。

といった点があります。

taichi-ishitani, 2023-02-06:
> リセットなし宣言の構文

* 変数宣言時にリセット不要であることを指示する
* リセット漏れチェックをしない `if_reset` の導入する
    * チェック漏れが出る恐れがありますが

はどうでしょうか？

dalance, 2023-02-06:
> 変数宣言時にリセット不要であることを指示する

これは良さそうなんですが、structやinterfaceのメンバー毎に指定ができないのが気になります。

taichi-ishitani, 2023-02-06:
> 例によって代入文は構文的には結構深い位置にある

パーサーの実装的にはあまり行儀が良いとは言えないでしょうが、`if_reset` の中に限りルールを、`代入文 | リセット不要指示` のようにするのはどうでしょうか？

dalance, 2023-02-06:
> パーサーの実装的にはあまり行儀が良いとは言えないでしょうが、if_reset の中に限りルールを代入文 | リセット不要指示 のようにするのはどうでしょうか？

文脈によって構文を切り替えるというのはかなり大きな判断ですね。
SVにおけるパースの難しさの主要因だと思っています。
正直ここまでの大変更（将来的な構文複雑性も含めて）を許容してまで導入すべき機能かどうか疑問に思ってはいます。

taichi-ishitani, 2023-02-06:
確かにご利益は少なそうですね。
リセット不要指示については、棄却でよいかと思います。

dalance, 2023-02-06:
ひとまず当初の要望であったリセット漏れチェックのみ追加しました。
リセット不要指示についてはいいアイデアが出てば再度検討したいと思います。

taichi-ishitani, 2023-02-07:
> #[not_reset] アトリビュート

Verilogの習慣の `// case Parallel` みたいに、`always_ff` とか `if_reset` に対してアトリビュートを付けるのはどうでしょうか？

dalance, 2023-02-07:
> Verilogの習慣の `// case Parallel` みたいに、`always_ff` とか `if_reset` に対してアトリビュートを付けるのはどうでしょうか？

リセット漏れチェックをそのalwaysだけ無効にする感じでしょうか？
lintをアトリビュートで無効化する機能があればそれで対応できそうですね。

taichi-ishitani, 2023-02-07:
> リセット漏れチェックをそのalwaysだけ無効にする感じでしょうか？

それでいいですし、チェック漏れを防ぐ観点からは、チェックを外す信号を個別に指定できる方が良いように思います。

dalance, 2023-02-07:
> それでいいですし、チェック漏れを防ぐ観点からは、チェックを外す信号を個別に指定できる方が良いように思います。

命名は適当ですがこんな感じでしょうか。

```
#[allow(not_reset("a", "b.d"))]
always_ff (clk, rst) {
    if_reset {
        c = 0;
    } else {
        a   = 0;
        b.d = 0;
        c   = 0;
    }
}
```

taichi-ishitani, 2023-02-07:
はい、このようなコードをイメージしていました。

dalance, 2023-02-07:
とりあえず allow アトリビュートを実装しました。
致命的なエラーでないものについては指定可能です。
missing_reset_statementの信号指定はまた別途考えます。

```
module Module29 (
    clk: input logic,
    rst: input logic,
) {
    var a: logic;
    var b: logic;
    #[allow(unused_variable)]
    var c: logic;

    #[allow(missing_reset_statement)]
    always_ff (clk, rst) {
        if_reset {
            a = 0;
        } else {
            a = 0;
            b = 0;
        }
    }

    #[allow(missing_port)]
    inst u0: Module29 (

    );
}
```

closed 2023-02-07 by dalance