veryl-lang/veryl #140: [ご相談] initial 文の対応について

ryuz, 2023-02-17:
いつもお世話になっております。
おそらくFPGA固有の話になってくるかと思うのですが、FPGAではリセットによる初期値とは別にブートROMなどからのビットストリームダウンロード時の初期値が存在し、しばし合成記述の中で変数の初期値であったり、initial 文での初期化によって指定する方法がとられます。
メモリ推論時などは initial 文の中で $readmemh で初期化したり、算術用のテーブルを for 文で作ってしまったりということもあります。
現状の Veryl には initial 文は無いかと思いますが対応される予定はありますでしょうか？

taichi-ishitani, 2023-02-18:
always_xxxは自身以外からの代入を禁止しているので、initial文と併用するなら、従来のalways文を使う必要があります。せっかくSV入でったチェック機構がなくなってしまうので、限定的に隠匿した形でinitial文を導入すべきなのではと思います。

なので、以下のように、always_ff内のみで使用できる制御構文（at_init）を導入して、この制御構文が含まれるalways_ff文は、従来のalways文とinitial文に分けてSVにするのが良いのではないかと思います。

```
always_ff {
　　at_init {
      a = 1;
    }
  a = b;
}
```

> 算術用のテーブルを for 文で作ってしまったりということもあります。

これについては、always_combでやるのが良いと思います。
シミュレーション開始時に必ず実行されるので、シミュレーションでも、テーブルの初期化が可能です。

taichi-ishitani, 2023-02-18:
変数の初期化であれば、宣言時の初期値代入でも良いような気もしますが、always_ffと併用出来るかは未確認です。

ryuz, 2023-02-18:
早速のご回答ありがとうございます。

> なので、以下のように、always_ff内のみで使用できる制御構文（at_init）を導入して、この制御構文が含まれるalways_ff文は、従来のalways文とinitial文に分けてSVにするのはどうでしょうか？

はい、何らかの形で SV 変換時に initial 文が生成できれば問題ないと考えます。

Xilinx 社 Vivado の話に限定かもしれませんが、シンセサイザのマニュアルにメモリの初期化は initial 文の中に書くように記載があり、現状これに従っております。頻度の少ない話ではありますので最悪そこだけ SV で書く手はあるのですが、Veryl にも何らかの記述方法があれば嬉しいです。

https://docs.xilinx.com/r/ja-JP/ug901-vivado-synthesis/HDL-%E3%82%BD%E3%83%BC%E3%82%B9-%E3%82%B3%E3%83%BC%E3%83%89%E3%81%A7-RAM-%E3%81%AE%E5%88%9D%E6%9C%9F%E5%86%85%E5%AE%B9%E3%82%92%E6%8C%87%E5%AE%9A

dalance, 2023-02-18:
`initial` 自体はパラメータチェックの `assert/assume` や `display` でも使いそうなので
普通に入れていいと思いましたがどうでしょうか。

どちらかというとかなり限定的な用途のために専用構文を割り当てる方が良くない気がします。
（専用構文で簡単に書けるということは使用を促進することになるので）

`always_ff` と `initial` を併用できないということであれば `always` を出力するアトリビュートなどあればいいかもしれません。

```
initial {
    a = 1;
}

#[emit(always)]
always_ff (i_clk, i_rst) {
    a = x;
}
```

taichi-ishitani, 2023-02-20:
上記の理由に加えて、「初期化はその FF に付随するので同じ always_ff 内の方が良かろう」との理由で、専用の制御構文を提案した次第です。
initial 文のサポートを決断したのであれば、特段の異論はありません。

DC では、initial 文での変数への代入はエラーになるか無視されるかなはずなので、シミュレーションと合成とで不一致が出てしまいます。
なので、initial 文での変数代入は、デフォルトではエラーにしておいた方が良いのではないかと思います。

dalance, 2023-02-20:
initial文をサポートするかどうかとFPGA向けのFF初期値をどのように記述するかは別に考えた方がいいかもしれませんね。
今のところ

* initial （finalも？）は入れても良さそう
* FPGAのFF初期値記述はもう少し考えた方が良さそう

という印象です。

taichi-ishitani, 2023-02-20:
> 変数の初期化であれば、宣言時の初期値代入でも良いような気もしますが、always_ffと併用出来るかは未確認です。

```systemverilog
module top;
  bit clk;
  bit a = 0;
  
  always_ff @(posedge clk) begin
    a <= ~a;
  end
  
  initial begin
    $finish;
  end
endmodule
```

上記の記述を各社のシミュレータでコンパイルしてみました。

* Aldec Riviera Pro
* Mentor QuestaSim
* Synopsys VCS

はコンパイルできましたが、

* Cadence Xcelium

はコンパイルエラーになりました。

dalance, 2023-02-21:
とりあえず `initial` / `final` は入れようと思います。
（`final` は必要性は低そうですがあえて排除する理由もなさそうなので）

initial内代入エラーの抑制も付けると、もともとの要望はこんな感じになるでしょうか。

```
#[allow(initial_assignment)]
initial {
    a = $readmemh("data.hex");
}

#[emit(always)]
always_ff (i_clk, i_rst) {
    a = x;
}
```

taichi-ishitani, 2023-02-21:
> #[allow(initial_assignment)]

合成段階での手戻りを防ぐためにも、アトリビュートの使用自体をグローバルに無効にする機能も必要ではないでしょうか？

dalance, 2023-02-21:
> 合成段階での手戻りを防ぐためにも、このアトリビュートの使用自体を、依存するプロジェクトも含めて、無効にする機能も必要ではないでしょうか？

確かにそうですが、そのようにモードを増やしていくと運用が難しくなりますね…。
（「ASICのときはA,Bをtrueに」「FPGAのときはX,Yをfalseに」といった具合に）

また仮に依存先でこのエラーが検出されたとして、その依存関係を使わないという以外にはどうしようもない
という点も気になります。

こういった特殊な記述はSVで書いてもらうことにして、Verylとしては対応しない方がいいかもしれません。

taichi-ishitani, 2023-02-21:
> 依存関係を使わない

確かにそうなのですが、早期に検知できれば、他のプロジェクトに差し替えるとか、特定のモジュールだけ自分で書くとか、対処はより容易なのかなと思います。

> そのようにモードを増やしていくと運用が難しくなりますね…。

* デフォルトは全機能有効
* 「使用禁止な機能リスト」を指定できるようにして、サブセットを定義できるようにする

様するのはどうでしょうか？

dalance, 2023-02-21:
> 確かにそうなのですが、早期に検知できれば、他のプロジェクトに差し替えるとか、特定のモジュールだけ自分で書くとか、対処はより容易なのかなと思います。

直接の依存関係であれば差し替え可能ですが、深い間接依存先でそのようなコードがあった場合
依存関係全体が使用不能になってしまいますし、
ただでさえ小さなエコシステムがモードによって細分化されていくのはあまりよくない気がしています。

taichi-ishitani, 2023-02-21:
となると、

> 特殊な記述はSVで書いてもらうことにして

になりますかね？

dalance, 2023-02-22:
例えばプロジェクト毎に `vendor = "xilinx"` くらいの粒度で指定するのは可能かもしれません。
（`vendor` 無指定のプロジェクトが `vendor = "xilinx"`のプロジェクトに依存するのは禁止みたいな感じで）

ryuz, 2023-02-22:
毎回面倒なご相談をしてしまい申し訳ないです。

> vendor = "xilinx" 

ベンダーは社名が変ったり(Xilinx -> AMD)、ツールバージョンでまたルールが変ったりすることも多いので、あまり良い手ではない気がしました。

個人的には Rust の std (OSあり)と no_std (OSなしのベアメタル)のように、言語としての使える機能がプロファイルとしてうまく住み分けできればなとは思いました。
とはいえ、Veryl の思想的に FPGA が合わない部分も多そうな気はしておりますので、「書けないものは SV で書く」というのはありかと思います。

dalance, 2023-02-22:
unsafeですかね。
プロジェクト毎にunsafeかどうかを設定して、unsafeでないプロジェクトはunsafeなプロジェクトに依存禁止。
unsafeプロジェクトではunsafeアトリビュートが使える、という感じで。
（アトリビュートの命名的にもこちらの方がいい気がします）

```
#[unsafe(initial_assignment)]
initial {
    a = $readmemh("data.hex");
}

#[unsafe(emit_always)]
always_ff (i_clk, i_rst) {
    a = x;
}
```

ryuz, 2023-02-23:
> unsafeですかね。

私は unsafe でも構わないのですが、Rust 的な感覚だと safe / unsafe はコードレビュー時のチェックリストの一部をコンパイラに任せてよい／悪い、を切り替えるポイントのイメージを持っているのでFPGAの為に使うにはもったいないワードな気もしました。
（将来の為に温存されても良いような予約語に思いました）

> プロジェクト毎にunsafeかどうかを設定して、unsafeでないプロジェクトはunsafeなプロジェクトに依存禁止。

Rust みたいに、使う側が 「unsafe とわかってって使います」と明示しないと使えない安全機構は良いと思いました。
ちなみにFPGAの場合、多分、 RAMとかFIFOとかのプリミティブなものでよく使うと思うので、結局全部 unsafe になりそうですね。

ryuz, 2023-02-23:
> プロジェクト毎にunsafeかどうかを設定して、unsafeでないプロジェクトはunsafeなプロジェクトに依存禁止。

ちょっと気になったのですが、この場合 Veryl から SV の module を使う場合も unsafe になるのでしょうか？
(Rust とかだと、C++ の関数を呼び出すところは unsafe だったかと思いますが)

taichi-ishitani, 2023-02-23:
対象のアトリビュートを使った時点でunsafeな訳ですから、プロジェクト内で使用を禁止するアトリビュートを指定できるようにできる方が良いのではないでしょうか？

ryuz, 2023-02-23:
Very の思想として ASIC 向けにデフォルト strict な言語であり、厳密な記述以外を含めない保証をすることで、言語で安全性を保障できる範囲を SV に対して広げたいというものかと想像しております。
 
一方で Rust がそうであるように FPGAなどでラフにラピッドプロトがやりたいケースでも生産性向上に寄与してくれそうに思っており、厳密性を緩めてほしいケースがありそうに思っておりご相談させて頂いている次第ですが、strict  の対義語を調べると lenient というのが出てきましたがこの手のワードを当てるとか、Rust の core と std みたいな位置づけとか如何でしょうか？

> unsafeなプロジェクトに依存禁止。

に対して、Rust で core なクレートは std なクレートに依存できない(逆は可) と似ている気がしています。

unsafe は、将来 ASIC 用に「ここだけは例外的な書き方をするけど、人間の方でしっかりケアしたから大丈夫(もしくはここはちゃんとレビューしないと危険というマーク)」に使えるように予約語を温存されてはいかがでしょうか？
Very 外のものとの接続など、厳密性を求める世界の中でも起こる例外の為に unsafe の語を温存しておいた方が有意義な気がしました。

dalance, 2023-02-23:
確かにunsafeもちょっと違いますね。
Rustのunsafeは「論理的には正しいけど（現在の技術では）コンパイラが証明できないので代わりに人間が保証する」というものなので
その人間を信用すれば安心して使うことができます。
一方で、今回の場合は「Vivadoの特定のバージョンでは正しいがそれ以外の環境では間違う」というものなのに
「Vivadoの特定のバージョン」という情報がどこにも書かれていないので、使っていいかどうかはコードを精査するまでわからなくなってしまいます。
（なので厳密にやるなら `target = "vivado@20XX.X"` のような指定が必要になりますが、それはVerylの領分を超えている気がします）

許可するアトリビュートを指定する方式にしても `initial_assignment` や `emit_always` というアトリビュートと「Vivado環境でRAMの初期値を `readmemh`  したい」というのの距離がありすぎて、何を指定していいかわからないです。
（今回の実装ではたまたまその2つでしたが、そうなるかどうかは実装次第ですし）

というわけで、今回のように特定の環境向けの特殊記述については（現時点では）SVで書くことにしておくのがいいような気がしてきました。

ryuz, 2023-02-23:
> というわけで、今回のように特定の環境向けの特殊記述については（現時点では）SVで書くことにしておくのがいいような気がしてきました。

ご検討ありがとうございました。 SV で書くという事でもすぐに困ることはないかと思います。

もし将来的に Veryl のオプション機能として縛りを緩める(SVで書ける事をいろいろ許容する)機能が追加されるようであれば、またご検討いただければ幸いです。

あと、一点、SV で書くのを前提とした場合、パッケージマネージャの方で GitHub にある SV のみの特定のタグのソースが取ってこれるとか、同じパッケージの中に Veryl と SV のソースコードを混在管理できるとかがあると便利に思いました。
もしよろしければご検討いただければ幸いです。

dalance, 2023-02-23:
> あと、一点、SV で書くのを前提とした場合、パッケージマネージャの方で GitHub にある SV のみの特定のタグのソースが取ってこれるとか、同じパッケージの中に Veryl と SV のソースコードを混在管理できるとかがあると便利に思いました。
もしよろしければご検討いただければ幸いです。

これも素朴に実装すると同じ問題が発生しそうですね…。
（依存関係を通してVivado専用のSVを引き込んでしまう可能性がある）
間接依存先からはSVは取ってこないとか何らかの制限は必要そうな気がします。

ryuz, 2023-02-23:
> これも素朴に実装すると同じ問題が発生しそうですね…。
>（依存関係を通してVivado専用のSVを引き込んでしまう可能性がある）
> 間接依存先からはSVは取ってこないとか何らかの制限は必要そうな気がします。

Vivado とは無関係に、既存の SV で書かれたモジュール資産を Veryl のソースからインスタンシエートして利用するということは少なからず必要なのかと思いますので、安全性の保証の問題はありそうですね。
既存の SV には何の仕組みも無いわけなので、取り込む側の Veryl 側から、取り込み先の URL と ハッシュに対してホワイトリストを与えるなり、何らかの「人間側で安全と確認したものしか取り込まない」仕組みが必要なように思いました。

taichi-ishitani, 2023-02-23:
依存先のプロジェクトの作者が、FPGA がメインで、ASIC 開発に明るくない場合、何が安全か確認のしようがないかと思います。
なので、使う側で確認するなりすべきで、

* 含まれる素の SV の一覧を取得できるようにして、安全性の確認をしやすくする
* 素の SV を上書きできる仕組みを用意する

が必要ではないでしょうか？

ryuz, 2023-02-23:
> 使う側で確認するなりすべきで、

これが基本なんだろうなと思いました。

> 素の SV を上書きできる仕組みを用意する

どこかに fork して、安全なように書き換えたもののコミットハッシュ値をホワイトリスト登録するのが再利用性が高そうにも思いましたが、たしかに Very 内で 中身を確認したり、上書きしたりできる仕組みがあると便利そうですね。

個人的には

1.  使う側で依存先のコードをしっかり確認して安全確認できたもののみ自プロジェクトのホワイトリストに加える
2.  ホワイトリストにないものは取り込まない

みたいな感じを想像していました。

dalance, 2023-02-23:
SVについては名前衝突の問題もありますね。
いわゆるdependency hell （複数の間接依存先で同パッケージ異バージョンが衝突する）を回避するためにVeryl側では衝突検知と名前の付け替えなどやっていますがSV側は無理です。
なのでSVは直接参照のみ、かつSVのインポートを明示する感じでしょうか。

```toml
"https://example.com/with_sv_prj" = {version = "0.1.0", import_sv = true}
```

これなら衝突回避はユーザの責任になりますし、
明示が必須なので実質ホワイトリスト相当になるかと思います。
ファイル単位などの細かい上書き制御は過剰かなという気がします。
自プロジェクトに取り込んで修正するか、folkして修正したものをインポートするかで対応可能かと。

オプションについてはsvとvlを一緒にインポートするのとsvだけインポートが必要かもしれません。
（後者は間接依存先のsvのみインポートしたい場合など）

dalance, 2023-02-24:
本件の解決にはなっていませんが、`initial` `final` は追加しました。
現時点では `display` くらいしかできることがないですが、将来的には `assume` などを行う想定です。

ryuz, 2023-02-24:
> 明示が必須なので実質ホワイトリスト相当になるかと思います。

依存先の依存先など階層の深いところで読み込まれているSVが、トップ階層から把握しきれなくなるケースが無いかと思った次第でした。

なので2階層以上深いところで依存している SV 関しては、一旦エラーにして、トップ階層で改めて許可(ホワイトリスト登録)して初めて通るような仕組みが必要かなと思いましたが、如何でしょうか？

dalance, 2023-02-24:
すいません。書き方が分かりにくかったかもですが、

> SVは直接参照のみ

が依存先の依存先など階層の深いところからSVを読み込まない（というより名前衝突の問題があるので読み込みたくても読み込めない）ということを言っています。
なので深いところのSVはホワイトリストを別途登録するのではなく、SVを読み込みたいものは別途直接依存に書くようにするというのが解決策になります。

例えば以下のようなprj1/prj2があって

```
prj1
  - src/a.vl
  - src/x.sv

prj2 (prj1に依存している)
  - src/b.vl
  - src/y.sv
```

以下のような依存関係を書くと

```toml
[dependencies]
"https://example.com/prj2" = "0.1.0"
```

以下のファイルが読み込まれます。

```
prj1/src/a.vl
prj2/src/b.vl
```

ここで

```toml
[dependencies]
"https://example.com/prj2" = {version = "0.1.0", import_sv = true}
```

とすると

```
prj1/src/a.vl
prj2/src/b.vl
prj2/src/y.sv
```

となります。

```toml
[dependencies]
"https://example.com/prj2" = {version = "0.1.0", import_sv = true}
"https://example.com/prj1" = {version = "0.1.0", import_sv_only = true}
```

とすると

```
prj1/src/a.vl
prj1/src/x.sv
prj2/src/b.vl
prj2/src/y.sv
```

となるイメージです。

ryuz, 2023-02-24:
> が依存先の依存先など階層の深いところからSVを読み込まない

なるほど、勘違いしておりました。すみません。

> というより名前衝突の問題があるので読み込みたくても読み込めない

こちらも了解です。つまりある程度 SV に追い出さないといけない機能を多用する場合、深い階層が取れなくなってくるということですね。

FPGAなどの場合、あまり小分けにせずに、モノリポ的に統合していった方がややこしい問題を起こさないかもしれませんね。

ryuz, 2023-03-12:
折角 initial 文をご対応いただいたので、下記の FPGA用 の SV のコードを Veryl で書き直せないか試し中です。
(汚いコードですみません)

https://github.com/ryuz/jelly/blob/master/rtl/math/jelly2_calc_crc.sv

CRCテーブルを initial で作成して ROM化しておいて、後からテーブル引きする構造なのですが、initial 文の中で代入を書こうとすると

```
assignment statement can't be placed at here
```

となってしまうので、上の方で一度話に出た  #[unsafe(initial_assignment)]  のようなものが無いとやはり難しそうですね。

やはりこの手のものは SV で書くのが無難そうです。

ちなみに、initial や always_comb の中でローカル変数を宣言したりするのも NG のようですが、このあたりもそういうポリシーで、必要なら function を作るべきという理解で良いでしょうか？

taichi-ishitani, 2023-03-12:
テーブルの初期化なら、always_combでも可能ですね。

taichi-ishitani, 2023-03-12:
論理的には違いはないと思うのですが、回路的に違いがあるのでしょうか？

ryuz, 2023-03-12:
> 論理的には違いはないと思うのですが、回路的に違いがあるのでしょうか？

試してないのですが、FPGAの内蔵SRAMブロックの初期化は少し特殊扱いなので、マニュアルにない方法で仮に今のバージョンで動いても将来保証されるものでもないので、どうしたものかなと思っております。

ryuz, 2023-03-12:
試しに今、テスト用に ROM を作って、初期化を always_comb  と initial とで切り替えてみたところ合成結果が変りました。

Vivado 2022.2 で、どちらも RAMは推論してくれたのですが、どうも initial だと中身で bit が固定な部分などがあると最適化しているようで少し小さな結果が出てきましたが  always_comb  だと少し大きいものになりました。
case 文で書いたテーブルを ROM推定してくれたりもするので、仮説ですが、前者はメモリ初期値として最適化していて、後者はロジックとして出した後に最適化の結果ROMを推論しているとか、何か解釈の違いは出ていそうな気がします。

dalance, 2023-03-13:
> ちなみに、initial や always_comb の中でローカル変数を宣言したりするのも NG のようですが、このあたりもそういうポリシーで、必要なら function を作るべきという理解で良いでしょうか？

特にこの件についてのポリシーというわけではありませんが、全体的なポリシーとして
「特に理由がない限りSystemVerilogを引き継ぐ」というのはあります。
基本的にSystemVerilog とセットで使う前提の言語なので、不要なところまで変えるべきではないということです。

なのでローカル変数の宣言についてはSystemVerilogがそうなっており、
特別にそれを許可する理由がないから、ということになります。

ryuz, 2023-03-13:
> なのでローカル変数の宣言についてはSystemVerilogがそうなっており、

勘違いしていたらすみません、現状の Veryl でも for 文でループ変数がローカル変数として for 内だけでつけると思いますが、SystemVerilog で begin の直後で変数が定義可能と思いますがそれはまた違うものなのでしょうか？

dalance, 2023-03-13:
> 勘違いしていたらすみません、現状の Veryl でも for 文でループ変数がローカル変数として for 内だけで使えると思いますが、SystemVerilog で begin の直後で変数が定義可能と思いますがそれはまた違うものなのでしょうか？

確かに `begin` の直後だけ書けますね…。
そうすると対応は可能になってきますが、
個人的には構文を複雑化させてまで追加するメリットは感じられないです。
実質的にalways_combでしか使えないですし。

taichi-ishitani, 2023-03-13:
```systemverilog
always_ff @(posedge i_clk) begin
  int temp;
  temp = a + b;
  c <= temp;
end
```

みたいに `always_ff` でも使えますね。
ただ、Veryl の場合、ノンブロッキング代入とブロックキング代入の字面上の違いがないので、混乱やバグの元になりそうです。
なので、always ブロック中での変数宣言は未サポートにして、中間変数が必要なぐらい複雑な記述は function などに切り出すようにしてもらうのが良いように思います。

ryuz, 2023-03-13:
まれに便利なケースがあるぐらいで、問題にならない気がしますので

> 中間変数が必要なぐらい複雑な記述は function などに切り出すようにしてもらうのが良いように思います。

に同意です。

ただ、逆に今の Veryl の function を試すと

- function ブロックの先頭以外でも var 宣言できてしまい、シミュレータに持っていくとエラーになる
- function であっても for や if などのブロック内だけのローカル変数は禁止の模様

という２点に気が付きました。後者は必要ならさらに function に分ければいいので不要と思いますが、前者は Veryl でもエラーになってくれる(もしくは自動でブロック先頭に移動してくれる)と嬉しいです。

dalance, 2023-03-13:
functionはブロック先頭に移動させるつもりで構文上は制限していなかったのですが未実装ですね…

ryuz, 2023-03-13:
> functionはブロック先頭に移動させるつもりで構文上は制限していなかったのですが未実装ですね…

個人的には変数はなるべく使う場所で宣言する方が好きですので、そういう方向性は嬉しいです。
特に今は困っておりませんので、将来ご検討いただければ幸いです。

dalance, 2023-03-14:
ちょっと話があちこちに飛んで、何が未検討なのか管理できなくなってきたので
残件を別issueにして一旦閉じます。
まとめると

* 発端となったinitialは追加済み
* FPGA用のRAM初期化はSVで書く
* SVの依存追加は #185 で継続
* functionの件は #184 で継続

です。もし見落としがあれば別issueを立ててもらえれば。

closed 2023-03-14 by dalance

dalance, 2024-11-13:
initial文での代入は #823 で再度検討しています。
今のアイデアはinitial_assignしたプロジェクトは`veryl publish`できない
（のでASICプロジェクトの依存関係に間違って入ってしまうことはない）というものです。