veryl-lang/veryl #141: Range syntax

dalance, 2023-02-21:
Range syntax is required to implement features related with `inside`.
In SystemVerilog, range is `[low_bound:high_bound]`.
This seems to be the same as part select syntax, but not equal. (ex, range operator `+:` `-:` can't be used)

So I think the range syntax should have a different syntax with part select.
For example, Rust style syntax:

* Veryl

```
assign x = a inside { 0..10, 20..=30 };
```

* Translated SystemVerilog

```verilog
assign x = a inside { [0:10-1], [20:30] };
```

This syntax has consistency with range specification of `for` statement.

```
for i: u32 in 0..10 {
   a = i;
}
```

The related issues: #11, #87, #124

taichi-ishitani, 2023-02-21:
To align characher size, how about using `...` instead of `..`?

taichi-ishitani, 2023-02-21:
Quartus does not support `inside` operator so I think Veryl should convert its `inside` operator to relational operators.

* 0...10
    * (a >= 0) && (a < 10)
* 20..=30
    * (a >= 20) && (a <= 30)

dalance, 2023-02-21:
> To align characher size, how about using ... instead of ..?

`..` and `...` have different meaning between each language.

* `0 <= a && a < 10`
    * Rust: `0..10`
    * Ruby: `0...10`
    * Swift: `0..<10`
    * Kotlin: `0 until 10`
* `0 <= a && a <= 10`
    * Rust: `0..=10`
    * Ruby: `0..10`
    * Swift: `0...10`
    * Kotlin: `0..10`

So  `0..<10` and `0..=10` may be most explicit.

taichi-ishitani, 2023-02-21:
> 0..<10 and 0..=10

It looks good for me.

dalance, 2023-02-21:
> Quartus does not support inside operator so I think Veryl should convert its inside operator to relational operators.

Expanding `inside` will reduce readability of SystemVerilog.
So I think the expansion should be enabled by option as the same as `implicit_parameter_types`.

dalance, 2023-02-22:
I read the discussion about range syntax of Rust https://github.com/rust-lang/rust/issues/28237.
The related opinions with this issue are below:

* about half-open range ( `0 <= a && a < 10` )
  * `..`
    * allmost all usage will be cover. So the shortest syntax is reasonable.
  * `..<`
    * there is syntax ambiguity if there is `0..`. ` 0 ..< 10` maybe ` (0..) < 10 ` or `0 (..<) 10`.
* about closed range ( `0 <= a && a <= 10` )
  * `...`
    * causes off-by-one error by mistype because `..` and `...` is very simillar.
  * `..=`
    * the meaning is confusable because it is simillar with assignment operator like `+=`.
  * `inclusive(0, 10)`
    * most explicit, but too long
* about other type range ( ex. `0 < a && a < 10` )
  * Is there any usecase?

Finally, Rust chose `..` and `..=`.
I think it may be suitable with Veryl. This is because

* Almost all case of `for` statement, half-open is used. `..<` is bit long for `for` statement
* `0..` can be mapped to `[0:$]` in SystemVerilog, so `0..` may be introduced in future.
* The concern about `...` is reasonable.
* The strangeness about `..=` can be acceptable because we already introduced `<:` and `>:`.

dalance, 2023-02-22:
> Quartus does not support inside operator so I think Veryl should convert its inside operator to relational operators.

Quartus has many unsupported features of SystemVerilog.

https://marekpikula.github.io/quartus-sv-gotchas/Intel%20Quartus%20SystemVerilog%20gotchas.html

So even if we add `inside` support for Quartus, Quartus may not be able to use actually.

dalance, 2023-03-01:
Range syntax is added.

closed 2023-03-01 by dalance