veryl-lang/veryl #143: [質問] ビット範囲の指定方法の経緯について

ryuz, 2023-02-23:
興味本位の質問で恐縮なのですが、Very の文法はかなり Rust 的な要素が多いと感じております。

その中で変数の宣言は

```
var a: logic<8>
```

のように MSB+1 を指定する Rust 的な指定なのに、特定ビット範囲の切り出し(スライス？)の指定は

```
assign b = a[7:0];
```

のようになっており、どうしてここは Rust 的な

```
assign b = a[0..8];
```

という MSB+1 を使う記法ではないのかなというのが少し気になりました。

もしよろしければ現状文法に至った経緯を教えて頂けないでしょうか？

すでにどこかで議論されていたら申し訳ありません。

taichi-ishitani, 2023-02-23:
範囲リテラルが未導入だったからとかでしょうか？

dalance, 2023-02-23:
範囲リテラルが未導入というだったのもありますが、
#141 に書いたように、range と part select は積極的に構文を分けた方がいいと思っているのもあります。
なのでpart selectに相当するRustの構文は存在しない（強いていうなら `as u32` みたいなやつかもしれませんが）のでSVを継承した感じです。

ryuz, 2023-02-23:
ご回答頂きありがとうございます。

> part selectに相当するRustの構文は存在しない（強いていうなら as u32 みたいなやつかもしれませんが）のでSVを継承した

なるほど、 range と part select の違いがよく理解できておりませんでしたので、理由が知れてすっきりしました。
有難うございます。

closed 2023-02-24 by dalance