veryl-lang/veryl #164: Integrated Test via Rust #[cfg(test)]

saturn77, 2023-03-05:
This is a great project, I really like the combination of Rust and hardware description language.

One question is would it be possible to have an in-built cycle accurate simulator to run tests - say for example in the code below, which is a basic Rust example, but the add_two and internal_adder would be Veryl code instead. 

The integration with verilator is nice, but if the test functionality could be "self-contained" that would be awesome.   

```rust 

pub fn add_two(a: i32) -> i32 {
    internal_adder(a, 2)
}

fn internal_adder(a: i32, b: i32) -> i32 {
    a + b
}

#[cfg(test)]
mod tests {
    use super::*;

    #[test]
    fn internal() {
        assert_eq!(4, internal_adder(2, 2));
    }
}
```

dalance, 2023-03-07:
I agree integrated unit test is useful.
I don't have a plan to develop built-in simulator now.
(In some distant future, maybe)

At the first step, I'll consider unit test syntax and code generation.
The second step is the integration with existing simulator (ex. Verilator, Vivado Simulator, VCS)

saturn77, 2023-03-07:
If the Veryl compiler could generate testbench code from Veryl code wrapped in  #[cfg(test]] macro that would be very good. 

If the Veryl code then ran that testbench code in Verilator or Vivado, that would automate the task of tool management and be similar to say what something like cooctb does. What would be great about the Veryl implementation of this is that all the code would be contained in one file - in the Veryl-Rust format. If additional Rust libraries could be exposed to the testbench, like an integrated DPI, this would also really make the  #[cfg(test)] macro extremely powerful. 

I imagine the integrated test code would look something like 

```rust
#[cfg(test)]
mod tests {
    use super::*;

    #[test]
    fn strobe_reset() {
        ... // check that the reset properly initializes all registers
    }
    
    #[test]
    fn run_main_code(){
        ... // run main code and assert on conditions, etc. 
    }

    #[test]
    fn run_random_stimulus(){
        ... // run main code but use random stimulus, and assert on conditions, etc. 
    }

    #[test]
    fn regression_all(){
        strobe_reset();
        run_main_code();
        run_random_stimulus();
    }      

}
```
Using Rust's log crate or tracing subscriber module would be a nice touch here to record the events of the test in addition to what Veryl or verilator might do. 

Then to run these tests you would simply have in your command list

```
veryl build 
veryl test --verilator 
```

closed 2024-04-15 by dalance