veryl-lang/veryl #165: Modport converse issue

saturn77, 2023-03-05:
This is really a feature request -- 

For the veryl code below, it would be great to a converse() function such that 

modport slave = converse ( master ); 

and it would be equivalent to what is below for the modport slave {...} definition. 

```veryl
interface InterfaceA #(
    parameter ParamA: u32 = 1,
    parameter ParamB: u32 = 1,
) {
    localparam ParamC: u32 = 1;

    var a: logic<ParamA>;
    var b: logic<ParamA>;
    var c: logic<ParamA>;

    // modport definition
    modport master {
        a: input ,
        b: input ,
        c: output,
    }

    modport slave {
        a: output ,
        b: output ,
        c: input ,
    }
}

module ModuleA (
    i_clk     : input logic               ,
    i_rst     : input logic               ,
    intf_a_mst: modport InterfaceA::master,
    intf_a_slv: modport InterfaceA::slave ,
) {
    // interface instantiation
    inst u_intf_a: InterfaceA [10];
}
```

dalance, 2023-03-06:
It looks good.
There are some syntax ideas.

* assignment

```
modport slave = converse(master);
```

This may be diverged from the original `modport` declaration.

* Rust style derive

```
#[converse(master)]
modport slave {}
```

This seems to be more consistent, but it may be confusable where `#[converse(master)]` can be added to.

* Rust style default syntax

```
modport slave {
    ..converse(master)
}
```

I think this is better than the others.
Other variation can be considered.

```
modport y {
    a: input,     // specify some ports
    ..converse(x) // other ports from another modport
}
modport x {
    ..input // all ports are input
}
```

saturn77, 2023-03-06:
I really like your different approaches, but also I would agree with the last one -- and I am using Rust markdown code fencing here so that there is some syntax highlighting . . . 

I think the type of declaration below is "Rusty" and elegant and compact. It also affords for flexibility on the port list, as the ...converse(x) can be a part of the modport, but not all of it! 

I think this is a very valuable feature. 

```rust 
modport y {
    a: input,     // specify some ports
    ..converse(x) // other ports from another modport
}
modport x {
    ..input // all ports are input
}
```

PR #1288 (2025-02-27): Support default member of modport
Closes #165
closed 2025-02-27 by dalance