veryl-lang/veryl #175: [質問] モジュール名の加工について

ryuz, 2023-03-12:
どこかにすでに書いてあったら恐縮なのですが、いくつか教えて頂けると助かります。

### モジュール名の加工について

現在の Veryl のソースを SV にするときに、モジュール名の前にプロジェクト名が付くようですが、例えば Rust の #[no_mangle] のように抑制する機能はあったりしますでしょうか？

### ファイル名も加工できないでしょうか？

モジュール名とファイル名が揃っていると、 verilator の `-y`  オプションや Vivado Simulator の `--sourcelibdir` オプションでの参照など、便利なことが多い気がするのですが、ファイル名もモジュール名と同様に加工することは可能でしょうか？

### バージョン番号も付与できたりしますでしょうか？

モジュール名にプロジェクト名だけでなく、バージョン番号もモジュール名に付与できるとSVで再配布するときも嬉しいことがありそうな気がしますが、如何でしょうか？

依存先の依存先のなどで、バージョン違いの同じパッケージが参照されていた場合なども共存できそうな気がしましたが、そのような事はありえますでしょうか？

dalance, 2023-03-13:
> 現在の Veryl のソースを SV にするときに、モジュール名の前にプロジェクト名が付くようですが、例えば Rust の #[no_mangle] のように抑制する機能はあったりしますでしょうか？

今のところありません。
ただ依存関係でない（つまりそのプロジェクトに直接所属している）ものはそもそもprefixの付与は不要なような気もします。
（prefix付与は依存関係間での衝突回避のためなので）

> モジュール名とファイル名が揃っていると、 verilator の -y オプションや Vivado Simulator の --sourcelibdir オプションでの参照など、便利なことが多い気がするのですが、ファイル名もモジュール名と同様に加工することは可能でしょうか？

どちらかというと上で書いたように名前を維持する方がいい気がします。
依存関係についてはファイル名の変更も可能ではありますが、そもそも生成されるファイルリストで参照すべきなので、あまりファイル名に依存すべきではないと思います。
（例えば参照先のバージョンが上がってファイルが増えた場合に、ファイル名を直接スクリプトに書いていると変更漏れが発生する、など）

> モジュール名にプロジェクト名だけでなく、バージョン番号もモジュール名に付与できるとSVで再配布するときも嬉しいことがありそうな気がしますが、如何でしょうか？

パッチバージョン変更が破壊的変更になってしまうなどデメリットの方が大きそうな気がします。
具体的なメリットはどのあたりでしょうか？

> 依存先の依存先のなどで、バージョン違いの同じパッケージが参照されていた場合なども共存できそうな気がしましたが、そのような事はありえますでしょうか？

現状でもバージョン違いのパッケージは参照可能です。
処理系側で名前衝突を検知して、自動で名前を付け替えています。
命名規則を仕様として露出してしまうと処理系側の自由度が減るのであまりうれしくないかもしれません。

ryuz, 2023-03-13:
> 現状でもバージョン違いのパッケージは参照可能です。
> 処理系側で名前衝突を検知して、自動で名前を付け替えています。

そのような機能があったのですね。気が付きませんでした。
それであればいろいろ余計なことを書いてしまった気がします。

> そもそもprefixの付与は不要なような気もします。

承知しました。将来変わる可能性もあるということですね。
現状、シミュレーション時に SV から、Veryl から変換した SV を飛び出しているので、SV からどう見えるのかを正しく理解しておきたいと思いました。
私も、衝突時以外名前が変わらないルールでも良いように思いました。

> そもそも生成されるファイルリストで参照すべきなので、あまりファイル名に依存すべきではないと思います。

承知いたしました。現状、シミュレーションや合成などを行うスクリプトの中から veryl build  を呼んでおりますので、その後  .f ファイルを取り込めばよい話ではあるかと思っています。
(ただ、パス名の変換など多少面倒だったので、横着がしたかったという理由です)

> パッチバージョン変更が破壊的変更になってしまうなどデメリットの方が大きそうな気がします。
> 具体的なメリットはどのあたりでしょうか？

Xilinx の IP コアジェネレータが バージョン番号入りのファイルをよく吐き出すので少し毒されていましたが、言われてみるとデメリットが多い気がしますね。
あえてメリットを考えるなら、パッケージマネージャが使えない環境化にSV化したファイルのみを流通させる場合で、呼ぶ側の素のSVと、Veryl由来のSVでバージョン整合を保証したい場合ぐらいでしょうか？
既に名前衝突時の対策があるとのことなので、不要な気がしております。

dalance, 2023-03-14:
Rustのように（FFIなど特殊用途を除けば）単体で完結している言語と異なり
VerylはSystemVerilogとの相互運用を前提としています。

なのでフローも既存のSystemVerilogプロジェクトに組み込むことになると思っていて
あまり特定のフローを押し付けない方がいいように思っています。
といっても標準的なフローは想定した方がいいかもしれないので、
そのあたりは実際に使いながら考えていくことになると思います。

ryuz, 2023-03-14:
回答ありがとうございます。状況よくわかりました。
製作者の意図と違う使い方をしているのではないかと不安になっていたのですが、引き続き試行錯誤しながら活用の仕方を模索させて頂きたいと思います。
ASIC開発ではないのであまり参考にはならないかもしれませんが、少し使った限りFPGA用途でも生産性を高めてくれそうに感じております。
引き続きよろしくお願いいたします。

closed 2024-01-05 by dalance