veryl-lang/veryl #176: [質問] キャストについてなど

ryuz, 2023-03-12:
Veryl 活用させて頂いております。 質問ばかりで恐縮なのですが。

### キャストについて

型のキャストについて

```
assign a = b as logic<12>;
assign c = d as u32;
```

などはできないのでしょうか？

localparam などで type を作れば問題ないのですが、int  にはキャストできたりするようなのでポリシーを知れると嬉しいです。

### 左辺での連接演算子

```
assign a = {b, c};
```

はできるようですが

```
assign {a, b} = c;
```

は、できないようですが、意図的に禁止されているものでしょうか？

dalance, 2023-03-13:
> 型のキャストについて

少なくとも `u32` はできるべきですね。
`logic` もできますが、 `logic<12>` は SystemVerilog の仕様上不可能です。

> int にはキャストできたりするようなのでポリシーを知れると嬉しいです。

intというのは組み込み型としては存在しないので、外部のユーザ定義型扱いで通っていると思います。

> 意図的に禁止されているものでしょうか？

意図的ではありませんでしたが、追加するとなると構文上の負荷は高そうです。
要検討かと思います。

ryuz, 2023-03-13:
ご回答ありがとうございます。

> logic もできますが、 logic<12> は SystemVerilog の仕様上不可能です。

承知いたしました(なんとなく想像はできます)。

Verilator の 幅チェックが割とうるさく、ビット幅変換によく bit 幅だけ指定したキャストを使っていて、使いたくなった次第です。
(特に bit 幅を増やす方向がきちんと書くと少し面倒ですので)。

回避手段はあるので、特に問題はありません。
理由が知れてすっきりしました、ありがとうございます。

taichi-ishitani, 2023-03-13:
```systemverilog
a = 4'(b);
```

みたいな、幅キャストは対応してましたっけ？

taichi-ishitani, 2023-03-14:
幅キャストが実装されれば、代入幅不一致問題は解決出来そうですね。

> a = b as 4;
> a = b as <4>;

どちらでもよいと思うのですが、後者ですと多次元配列へのキャストもできそうに見えてしまうのが、少々難点ではあります（大した問題にはならないと思いますが）。

dalance, 2023-03-14:
> どちらでもよいと思うのですが、後者ですと多次元配列へのキャストもできそうに見えてしまうのが、少々難点ではあります（大した問題にはならないと思いますが）。

前者だとなんのキャストかよくわからない気がしたので後者かなと思いましたが、多次元指定できてしまうのは確かにそうですね。

dalance, 2023-03-14:
というかSystemVerilogのキャストが普通の意味の型キャストではなくて、もっといろいろな変換ができてしまう感じなのですよね。
ちょっと全体的に整合性が取れるよう考え直した方がいいかもしれません。

taichi-ishitani, 2023-03-14:
> 多次元配列へのキャスト

SystemVerilog 的には、packed であれば型が何であれ代入は可能なので、

```
a = b as <2>;
c = d as <2, 3>;
e = f as foo_struct;
g = h as foo_struct <2, 3>
```

のようなコードを、

```systemverilog
a = (2)'(b);
c = (2*3)'(d);
e = foo_struct'(f);
g = ($bits(foo_struct)*2*3)'(h);
```

の様に変換するのはありかもですね。

dalance, 2024-04-16:
組み込み型へのキャストについては#499 にあるのでこちらはクローズします。

closed 2024-04-16 by dalance