veryl-lang/veryl #178: [Feature] User-Defined Synthesizable Generic Types

BinhaoQin, 2023-03-13:
As far as I know, `struct` in system verilog cannot have type parameter. Although interface could have one, the constraint on the type parameter cannot be expressed when defining a module as interfaces get their type parameter where they instantiate, not where they are required. Therefore, I would like to have something similar to a struct but can be parameterized in terms of type. I understand this is not a verilog feature, rather it requires doing codegen before transplile. The following is an example.
```veryl
#[repr(packed)]
struct Option<T: type> {
    logic<1> valid;
    T value;
}
```

dalance, 2023-03-14:
I'm interesting in it.
I think generics which has not been provided in SystemVerilog will be useful.
The following issus has the same purpose.

#58
#39

So I'll consider consistent generics syntax through `struct` `function` `package` ( maybe `module`?) .

dalance, 2024-01-05:
Move to #191

closed 2024-01-05 by dalance