veryl-lang/veryl #179: [Feature] Rust Style Generic Syntax

BinhaoQin, 2023-03-13:
Basically, instead of `module A #(parameter I) ();`, we can have `module A<I: parameter>();`

dalance, 2023-03-14:
I compared the current syntax and Rust style.

* current syntax

```
module A #(
    parameter  a: u32 = 1,
    localparam b: u32 = 1,
) {
    localparam c: u32 = 1;
}

module B {
    inst u: A #(
        a: 1,
    ) ();
}
```

* Rust style

```
module A <
    a: parameter  u32 = 1,
    b: localparam u32 = 1,
> {
    localparam c: u32 = 1;
}

module B {
    inst u: A <
        a: 1,
    > ();
}
```

I think localparam declaration of the current is more consistent with in-module localparam declaration than Rust style.
`#()` and `<>` is not so different. If there is not advantage of `<>`, `#()` may be better because it is familiar with SystemVerilog user.
I think `<>` may be useful for Veryl's generics #178 .

dalance, 2024-01-05:
Move to #191

closed 2024-01-05 by dalance