veryl-lang/veryl #186: Auto-documentation

saturn77, 2023-03-14:
This is a language feature request, not an issue. 

Would it be possible to documentation to automatically create an md-book something like
as below -- using the '//!' escape clause as shown below? 

The TerosHDL available in VSCode does something similar, I use it to automatically create documentation embedded in the SystemVerilog Code. 

For example, all variables get at Table listing, and port list, and processes, and state machines, etc.

```
 //!  Variables 
    var r_data0: logic<ParamA>;     //! r_data0 is for the button 
    var r_data1: logic<ParamA>;     //! r_data1 is for debounced version of button 
```

dalance, 2023-03-20:
Looks good.
I'll implement it at #195.

The current implementation is like below:

```
/// Test module for doc comment
///
/// * list item0
/// * list item1
module Module36 #(
    /// Data width
    parameter  ParamA: u32 = 1,
    localparam ParamB: u32 = 1,
) (
    i_clk  : input  logic        , /// Clock
    i_rst_n: input  logic        , /// Reset
    i_data : input  logic<ParamA>, /// Data input
    o_data : output logic<ParamA>, /// Data output
) {}

/// Test interface for doc comment
///
/// * list item0
/// * list item1
interface Interface36 #(
    parameter  ParamA: u32 = 1, /// Data width
    localparam ParamB: u32 = 1,
) {}

/// Test package for doc comment
///
/// * list item0
/// * list item1
package Package36 {}
```

The rendered HTML is

[image]

saturn77, 2023-03-20:
> image

This will add a great feature to the language, documentation is so 
important.

PR #195 (2023-03-20): Automatic ducumentation
Closes #186
closed 2023-03-20 by dalance