veryl-lang/veryl #324: [Bug] Packages are mistranslated to systemverilog

nblei, 2023-07-28:
Veryl's build prefixes the project name to each SV package name.

Thus, veryl package `a` in project `p` becomes `package p_a` in SV.

However, veryl does not translate the name of the package when it is used in other SV modules.

A simple example:
```
#!/usr/bin/bash
veryl new pkgbug
cd pkgbug
mkdir src
mkdir pkg
echo "package my_pkg {localparam I: u32 = 12;}" >pkg/my_pkg.vl
echo "module my_mod {var _k: u32 = my_pkg::I; }" >src/my_mod.vl
veryl build
 ```
 
This creates: `my_pkg.sv`
```
package pkgbug_my_pkg;
    localparam int unsigned I = 12;
endpackage
```
and `my_mod.sv`
```
module pkgbug_my_mod;
    int unsigned _k;
    assign _k = my_pkg::I;
endmodule
```

dalance, 2023-07-31:
In this case, I think the following output is expected.

```
module pkgbug_my_mod;
    int unsigned _k;
    assign _k = pkgbug_my_pkg::I;
endmodule
```

But the current syntax may be ambiguous.
If `my_pkg` refers an external SystemVerilog package, the prefix is not required.
So explicit SystemVerilog namespace syntax may be required like below.

```
module my_mod {
  var _k: u32 = my_pkg::I;
  var _i: u32 = $sv::my_pkg::I;
}
```

```
module pkgbug_my_mod;
    int unsigned _k;
    assign _k = pkgbug_my_pkg::I;
    int unsigned _i;
    assign _i = my_pkg::I;
endmodule
```

dalance, 2023-07-31:
I fixed this issue through #329.
Could you try it?

closed 2023-07-31 by dalance

nblei, 2023-07-31:
Working for my use case. Thanks, Dalance.

PR #329 (2023-07-31): Fix scoped identifier translation
Closes #324