veryl-lang/veryl #339: ASPLOS Tutorial

nblei, 2023-08-04:
Hey Dalance, what do you think of the idea of doing a tutorial or workshop at ASPLOS (https://www.asplos-conference.org/asplos2024/cfp/) or similar architecture / EDA conference?

 Personally I think Veryl has the opportunity to replace SV as an HDL and that there will be a good amount of interest in Veryl generated by a tutorial.

dalance, 2023-08-07:
Thank you for your suggestion. It seems to be good idea.
But I don't have an experience about such conference, so it seems to be difficult...

nblei, 2023-08-07:
I can handle that.

I will have quite a busy fall/winter, so probably won't look to present a tutorial at ASPLOS, but perhaps ISCA or DAC, both are in June.  That also gives a good amount of time to get Veryl into a nicely polished state.

dalance, 2023-08-10:
Thanks.
I'll consider to submit to such workshop at the next chance.