veryl-lang/veryl #347: [Feature] Support for asynchronous set signal

nblei, 2023-08-07:
Modern standard cell  libraries include flops with asynchronous sets.  For example, DC maps

```
module f(
    input clk_i,
    input rst_ni,
    input set_i,
    input q_i,
    output reg q_o
);

always_ff @(posedge clk_i or negedge rst_ni or negedge set_i)
    if (!rst_ni) q_o <= 1'b0;
    else if (!set_i) q_o <= 1'b1;
    else q_o <= q_i;

endmodule
```
to
```

module f ( clk_i, rst_ni, set_i, q_i, q_o );
  input clk_i, rst_ni, set_i, q_i;
  output q_o;

  // I have redacted the specific register name to avoid any NDA violations
  reg_type1 q_o_reg ( .d(q_i), .clk(clk_i), .rb(rst_ni), .psb(set_i), 
        .o(q_o) );
endmodule
```
And removing the reset signal from the verilog, we get
```
module counter ( clk_i, rst_ni, set_i, q_i, q_o );
  input clk_i, rst_ni, set_i, q_i;
  output q_o;

  reg_type_2 q_o_reg ( .d(q_i), .clk(clk_i), .psb(set_i), .o(q_o) );
endmodule
```

Thus, Veryl needs to support a `set` signal in `always_ff` declarations.  I propose the following:

1: Redefine the `AlwaysFfDeclaration` to have two optional `AlwaysFfSetReset` non-terminals (which is just a renamed `AlwaysFfReset`.
```
AlwaysFfDeclaration: AlwaysFf LParen AlwaysFfClock [Comma AlwaysFfSetReset] [Comma AlwaysFfSetReset] RParen LBrace { Statement } RBrace;
```
2: Assume that if two `AlwaysFfSetReset` non-terminals are present, then the order is `set` then `reset` (largely because 'set-reset' sounds better in english than 'reset-set').

3: Add a `IfSetStatement` statement.

4: While flip-flops typically prioritize reset > set, it is easy enough to prioritize set > reset with two inverters.  Thus, let the user specify their intended priority based on the order of the `IfSetStatement` and `IfResetStatement` statements.

Veryl is feature incomplete if it cannot handle asynchronous set flip-flops.

dalance, 2023-08-09:
I don't have an experience to write FF with async set and reset through `always_ff` declaration.
Some people pointed out that such description causes mismatch between simulation and synthesis.

https://stackoverflow.com/questions/18641159/can-systemverilog-represent-a-flip-flop-with-asynchronous-set-and-reset-without
https://blog.yosyshq.com/p/async-load-ff/

According to IEEE Std 1800-2017 10.6.1, `assign` and `deassign` are required to model FF with async set and reset like below:

```
module dff (q, d, clear, preset, clock);
    output q;
    input d, clear, preset, clock;
    logic q;
    always @(clear or preset)
        if (!clear)
            assign q = 0;
        else if (!preset)
            assign q = 1;
        else
            deassign q;
    always @(posedge clock)
        q = d;
endmodule
```

Additionally, such usage of `deassign` seems to be considered to deprecete according to C.4.2 of Std.

Therefore, I think that instantiating such special cell may be better than inferring through synthesis.
What do you think?

dalance, 2023-08-10:
I think splitting file to for synthesis and simulation may increase file management cost.
So using `ifndef SYNTHESIS` may be better than splitting file.

```
always_ff @(posedge i_clk or negedge i_rst_n or negedge i_set0)
    if (!i_rst_n) o_q0 <= 1'b0;
    else if (!i_set0) o_q0 <= 1'b1;
    else o_q0 <= i_d0;

`ifndef SYNTHESIS
always @(i_rst_n or i_set0)
    if (i_rst_n && !i_set0) force o_q0 = 1'b1;
    else release o_q0;
`endif
```

I thought about syntax and semantics.
I think that the meaning of `set` and `reset` is not so clear.
Even if we define that `reset` is  0 and `set` is 1, the following case is not clear.

```
// Sometimes we want to reset by 1.
logic o_q0;
always_ff @(posedge i_clk or negedge i_rst_n)
    if (!i_rst_n) o_q0 <= 1'b1;
    else o_q0 <= i_d0;

// 'haa is reset value and 'hff is set value, or reversed?
logic [7:0] o_q1;
always_ff @(posedge i_clk or negedge i_rst_n or negedge i_set0)
    if (!i_rst_n) o_q1 <= 8'haa;
    else if (!i_set0) o_q1 <= 8'hff;
    else o_q1 <= i_d1;
```

So I think that FF with multiple resets should be added instead of FF with set and reset like below.

```
// Veryl code

// FF with reset
always_ff (i_clk, i_rst_n) {
    // argument of if_reset can be omitted.
    if_reset {
    } else {
    }
}

// FF with set and reset
always_ff (i_clk, i_rst_n, i_set) {
    // explicit specification of reset signal
    if_reset i_rst_n {
    } else if_reset i_set {
    } else {
    }
}

// more resets can be used
always_ff (i_clk, i_set0, i_set1, i_set2) {
    if_reset i_set0 {
    } else if_reset i_set1 {
    } else if_reset i_set2 {
    } else {
    }
}
```

nblei, 2023-08-10:
Say for example that you have an 8-bit register which, at reset, should be 8'bF0.

`always_ff @(posedge clk or negedge rst) if (!rst) q <= 8'bF0; else q <= d;`

Assuming a robust sequential cell library, synthesis and mapping will infer two different types of flip flops for the two nibbles.

For your example with `i_set0` through `i_set2`, I am unfamiliar with any flip-flop which has three asynchronous resets, meaning that will only be synthesizable if we assume at least one signal is synchronous.  It also makes the intended semantics unclear.  As you brought up, the semantics of a JK-flip-flop don't actually match the semantics of the verilog used to infer them, hence the need to use `force` and `release` and other typically non-synthesizable constructs in order to make sure simulation semantics match hardware semantics.  So in the case of two asynchronous reset signals, it is clear what Veryl's semantics should be.

I'll also note that, typically, flip-flops are bistable --- so what exactly is the point of `i_set2`?  Into what state does `i_set2` reset the flip-flop that `i_set0` or `i_set1` hasn't already covered?

dalance, 2023-08-10:
I think that 3 or more reset semantics is clear.
(Of course, I've never write such RTL in production...)

```
always_ff @(posedge i_clk or negedge i_set0 or negedge i_set1 or negedge i_set2 or negedge i_set3)
    if (!i_set0) o_q0 <= 1'b0;
    else if (!i_set1) o_q0 <= 1'b1;
    else if (!i_set2) o_q0 <= 1'b0;
    else if (!i_set3) o_q0 <= 1'b1;
    else o_q0 <= i_d0;
```

It will be synthesized to below:

```
wire rb = !(!i_set0 | (i_set0 & i_set1 & !i_set2));
wire psb = !((i_set0 & !i_set1) | (i_set0 & i_set1 & i_set2 & !i_set3));
reg_type1 q_o_reg ( .d(q_i), .clk(clk_i), .rb(rb), .psb(psb), .o(q_o) );
```

I summarized my idea.

* In RTL, set and reset are not distinguished. Instead of it, we specify a reset value. Synthesizer will decide which pin (rb or psb) should be used according to the reset value.
* So introducing async set and reset to Veryl means supporting two async resets with the different values.
* If two resets are allowed, there is no reason that more resets are forbidden.

nblei, 2023-08-16:
Makes sense to me.  This is then clear about what hardware it is meant to synthesize, emits SystemVerilog which matches the hardware semantics for simulation.