veryl-lang/veryl #438: Back annotations

nblei, 2023-12-13:
Dalance,

Not sure if there's interest, but a nice feature may be emitting SystemVerilog with back annotations of some sort.
While this would help humans who are performing verification to identify links between emitted SV and Veryl, I think the bigger benefit here would be in enabling tools to automatically associate the emitted SV with the generating Veryl code.  Consider, for example, a SV LSP which allows you to GOTO a veryl file, or a synthesis or APR script which can report timing/area/power/thermal violations at the Veryl level.

dalance, 2023-12-28:
I'm interesting in this feature.
One of the dissatisfied points of Chisel was missing machine readable source map.

The consideration points of this feature:

* Where are there annotations?
  * Inline of the generated source code
  * Separated files corresponding to the generated source codes
  * Single file
* What format does the annotations have?
  * text format ( ex. TOML/JSON/...)
  * binary format

The sourcemap of JavaScript seems to be useful as reference.

https://sourcemaps.info/spec.html
https://www.bugsnag.com/blog/source-maps/

At the moment, I prefer "Single file" and "TOML format" with Base 64 VLQ encoded entries.

dalance, 2024-05-29:
I'm tring to add source map support at #741.
I adopted the source map format of JavaScript defined at https://sourcemaps.info/spec.html.
This is because the format has large ecosystem in JS and Rust world.

For example, the following link is the result of `delay.veryl` in std by [Source Map Visualization](https://evanw.github.io/source-map-visualization/).

https://evanw.github.io/source-map-visualization/#OTY0AC8vLyBEZWxheSBpbnB1dCBieSBjb25maWd1cmVkIGN5Y2xlCm1vZHVsZSB2ZXJ5bF90ZXN0Y2FzZV9kZWxheSAjKAogICAgLy8vIENsb2NrIGN5Y2xlIG9mIGRlbGF5CiAgICBwYXJhbWV0ZXIgaW50IHVuc2lnbmVkIERFTEFZID0gMSwKICAgIC8vLyBJbnB1dC9vdXRwdXQgZGF0YSB3aWR0aAogICAgcGFyYW1ldGVyIGludCB1bnNpZ25lZCBXSURUSCA9IDEsCiAgICAvLy8gSW5wdXQvb3V0cHV0IGRhdGEgdHlwZQogICAgcGFyYW1ldGVyIHR5cGUgVFlQRSA9IGxvZ2ljIFtXSURUSC0xOjBdCikgKAogICAgLy8vIENsb2NrCiAgICBpbnB1dCBsb2dpYyBpX2NsaywKICAgIC8vLyBSZXNldAogICAgaW5wdXQgbG9naWMgaV9yc3RfbiwKICAgIC8vLyBJbnB1dAogICAgaW5wdXQgVFlQRSBpX2QsCiAgICAvLy8gT3V0cHV0CiAgICBvdXRwdXQgVFlQRSBvX2QKKTsKICAgIGlmICgoREVMQVkgPj0gMSkpIGJlZ2luIDpnX2RlbGF5CiAgICAgICAgVFlQRSBkZWxheSBbMDpERUxBWS0xXTsKCiAgICAgICAgYWx3YXlzX2NvbWIgb19kID0gZGVsYXlbREVMQVkgLSAxXTsKICAgICAgICBhbHdheXNfZmYgQCAocG9zZWRnZSBpX2NsaywgbmVnZWRnZSBpX3JzdF9uKSBiZWdpbgogICAgICAgICAgICBpZiAoIWlfcnN0X24pIGJlZ2luCiAgICAgICAgICAgICAgICBkZWxheSA8PSAnezB9OwogICAgICAgICAgICBlbmQgZWxzZSBiZWdpbgogICAgICAgICAgICAgICAgZGVsYXlbMF0gPD0gaV9kOwogICAgICAgICAgICAgICAgZm9yIChpbnQgdW5zaWduZWQgaSA9IDE7IGkgPCBERUxBWTsgaSsrKSBiZWdpbgogICAgICAgICAgICAgICAgICAgIGRlbGF5W2ldIDw9IGRlbGF5W2kgLSAxXTsKICAgICAgICAgICAgICAgIGVuZAogICAgICAgICAgICBlbmQKICAgICAgICBlbmQKICAgIGVuZCBlbHNlIGJlZ2luIDpnX25vX2RlbGF5CiAgICAgICAgYWx3YXlzX2NvbWIgb19kID0gaV9kOwogICAgZW5kCmVuZG1vZHVsZQoyMTIyAHsidmVyc2lvbiI6MywiZmlsZSI6ImRlbGF5LnN2Lm1hcCIsInNvdXJjZXMiOlsiLi4vLi4vc3RkL3NyYy9kZWxheS9kZWxheS52ZXJ5bCJdLCJzb3VyY2VzQ29udGVudCI6WyIvLy8gRGVsYXkgaW5wdXQgYnkgY29uZmlndXJlZCBjeWNsZVxucHViIG1vZHVsZSBkZWxheSAjKFxuICAgIC8vLyBDbG9jayBjeWNsZSBvZiBkZWxheVxuICAgIHBhcmFtIERFTEFZOiB1MzIgPSAxLFxuICAgIC8vLyBJbnB1dC9vdXRwdXQgZGF0YSB3aWR0aFxuICAgIHBhcmFtIFdJRFRIOiB1MzIgPSAxLFxuICAgIC8vLyBJbnB1dC9vdXRwdXQgZGF0YSB0eXBlXG4gICAgcGFyYW0gVFlQRTogdHlwZSA9IGxvZ2ljPFdJRFRIPixcbikgKFxuICAgIC8vLyBDbG9ja1xuICAgIGlfY2xrOiBpbnB1dCBjbG9jayxcbiAgICAvLy8gUmVzZXRcbiAgICBpX3JzdDogaW5wdXQgcmVzZXQsXG4gICAgLy8vIElucHV0XG4gICAgaV9kOiBpbnB1dCBUWVBFLFxuICAgIC8vLyBPdXRwdXRcbiAgICBvX2Q6IG91dHB1dCBUWVBFLFxuKSB7XG4gICAgaWYgKERFTEFZID49IDEpIDpnX2RlbGF5IHtcbiAgICAgICAgdmFyIGRlbGF5OiBUWVBFIFtERUxBWV07XG5cbiAgICAgICAgYXNzaWduIG9fZCA9IGRlbGF5W0RFTEFZIC0gMV07XG4gICAgICAgIGFsd2F5c19mZiAoaV9jbGssIGlfcnN0KSB7XG4gICAgICAgICAgICBpZl9yZXNldCB7XG4gICAgICAgICAgICAgICAgZGVsYXkgPSAnezB9O1xuICAgICAgICAgICAgfSBlbHNlIHtcbiAgICAgICAgICAgICAgICBkZWxheVswXSA9IGlfZDtcbiAgICAgICAgICAgICAgICBmb3IgaTogdTMyIGluIDEuLkRFTEFZIHtcbiAgICAgICAgICAgICAgICAgICAgZGVsYXlbaV0gPSBkZWxheVtpIC0gMV07XG4gICAgICAgICAgICAgICAgfVxuICAgICAgICAgICAgfVxuICAgICAgICB9XG4gICAgfSBlbHNlIDpnX25vX2RlbGF5IHtcbiAgICAgICAgYXNzaWduIG9fZCA9IGlfZDtcbiAgICB9XG59XG4iXSwibmFtZXMiOlsiIiwibW9kdWxlIiwiZGVsYXkiLCIjIiwiKCIsInBhcmFtZXRlciIsImludCB1bnNpZ25lZCIsIkRFTEFZIiwiPSIsIjEiLCIsIiwiV0lEVEgiLCJ0eXBlIiwiVFlQRSIsImxvZ2ljIiwiWyIsIl0iLCIpIiwiaW5wdXQiLCJpX2NsayIsImlfcnN0X24iLCJpX2QiLCJvdXRwdXQiLCJvX2QiLCI7IiwiaWYiLCI+PSIsIjoiLCJnX2RlbGF5IiwiYWx3YXlzX2NvbWIiLCItIiwiYWx3YXlzX2ZmIiwiYmVnaW4iLCIneyIsIjAiLCJ9IiwiZW5kIiwiZWxzZSIsImZvciIsImkiLCJnX25vX2RlbGF5IiwiZW5kbW9kdWxlIl0sIm1hcHBpbmdzIjoiQUFBQUE7QUFDSUMsc0JBQU9DLE1BQU1DLENBQUNDOztJQUVkQyxVQUFhQyxhQUFQQyxNQUFXQyxFQUFFQyxDQUFDQzs7SUFFcEJMLFVBQWFDLGFBQVBLLE1BQVdILEVBQUVDLENBQUNDOztJQUVwQkwsVUFBWU8sS0FBTkMsS0FBV0wsRUFBRU0sTUFBS0MsQ0FBQ0osU0FBS0ssQ0FBQ2hCO0FBQ25DaUIsRUFBRWI7O0lBRVNjLE1BQU1KLE1BQWJLLEtBQWtCVDs7SUFFWFEsTUFBTUosTUFBYk0sT0FBa0JWOztJQUViUSxNQUFNTCxLQUFYUSxHQUFlWDs7SUFFVlksT0FBT1QsS0FBWlUsR0FBZ0J2QjtBQUNwQmlCLENBQUVPO0lBQ0VDLElBQUdyQixDQUFDRyxNQUFNbUIsR0FBR2pCLENBQUNRLFNBQUVVLENBQUNDLE9BQVE1QjtRQUNWYSxLQUFQWCxNQUFZYSxHQUFDUixPQUFLUyxDQUFDUTs7UUFFdkJLLFlBQU9OLElBQUlmLEVBQUVOLEtBQUthLENBQUNSLE1BQU11QixFQUFFckIsQ0FBQ08sQ0FBQ1E7UUFDN0JPLFlBQVUzQixTQUFDZSxLQUFLVCxVQUFFVSxPQUFLSCxFQUFFZTtZQUNyQlAsY0FBU087Z0JBQ0w5QixPQUFNTSxFQUFFeUIsRUFBRUMsQ0FBQ0MsQ0FBQ1g7WUFDaEJZLElBQUVDLEtBQUtMO2dCQUNIOUIsS0FBS2EsQ0FBQ21CLENBQUNsQixHQUFFUixFQUFFYSxHQUFHRztnQkFDZGMsS0FBT2hDLGFBQUhpQyxJQUFVOUIsR0FBVjhCLElBQWFoQyxPQUFiZ0MsS0FBbUJQO29CQUNuQjlCLEtBQUthLENBQUN3QixDQUFDdkIsR0FBRVIsRUFBRU4sS0FBS2EsQ0FBQ3dCLEVBQUVULEVBQUVyQixDQUFDTyxDQUFDUTtnQkFDM0JZO1lBQ0pBO1FBQ0pBO0lBQ0pBLElBQUVDLFdBQUtWLENBQUNhLFVBQVd4QztRQUNmNkIsWUFBT04sSUFBSWYsRUFBRWEsR0FBR0c7SUFDcEJZO0FBQ0pLIn0=

dalance, 2024-06-07:
I added a feature to remap `path:line:col` in verilator's log at `veryl test`.
Now we can see where is root cause in Veryl's code easily.

```
%Error: /home/hatta/work/repos/veryl/testcases/sv/48_test.sv:11: Verilog $stop
        ^ from: /home/hatta/work/repos/veryl/testcases/veryl/48_test.veryl:4:18
[ERROR]       Failed test (test1)
[INFO ]    Compiling test (test2)
%Error: /home/hatta/work/repos/veryl/testcases/sv/48_test.sv:23:1: syntax error, unexpected endmodule
        ^ from: /home/hatta/work/repos/veryl/testcases/veryl/48_test.veryl:18:18
   23 | endmodule
      | ^~~~~~~~~
%Error: Cannot continue
[ERROR]       Failed compile (test2)
[INFO ]    Compiling test (test3)
%Error-PROCASSWIRE: /home/hatta/work/repos/veryl/testcases/sv/48_test.sv:31:17: Procedural assignment to wire, perhaps intended var (IEEE 1800-2023 6.5): 'a'
                    ^ from: /home/hatta/work/repos/veryl/testcases/veryl/48_test.veryl:26:18
                                                                              : ... note: In instance 'test3'
   31 |     always_comb a = 1;
      |                 ^
                    ... For error description see https://verilator.org/warn/PROCASSWIRE?v=5.024
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

nblei, 2024-06-07:
This is beautiful, and a fantastic example of why open source hardware
development tools have a place.

dalance, 2024-06-11:
I'll merge #741, and this issue will be closed.
The remaining features are moved to #776 because I think the feature to consume source map should be separated from Veryl compiler.

PR #741 (2024-06-11): Add source map generation support
Closes #438
closed 2024-06-11 by dalance