veryl-lang/veryl #499: Cast to builtin types

dalance, 2024-02-09:
Now cast to builtin types will be syntax error.

```
Expression11: Expression12 { As ScopedIdentifier };
```

The simple way to support it:

```
Expression11: Expression12 { As ScalarType };
```

But it will generate the following code

```verilog
assign a = int signed'(x);
```

It is invalid SystemVerilog.

taichi-ishitani, 2024-03-31:
How about this workaround?

* Reserve identifiers starting two udnerscore `__` as identifiers used by Veryl
* If there is an expression like `foo as u32` then Veryl inserts `typedef int unsigned __u32` and convert this Veryl expression to `__u32'(foo)`

dalance, 2024-08-15:
The following cast is valid, so #884 implements it.

```systemverilog
assign a = signed'(int'(x));
```

PR #884 (2024-08-15): Support cast to builtin types
Closes #499
closed 2024-08-15 by dalance