veryl-lang/veryl #528: Built-in syntax for bit-width calculation

dalance, 2024-03-07:
In the caluculation of bit width, `(W > 1) ? $clog2(W) : 1` pattern is frequent.
This because if `W` is 1, `$clog2(W)` return 0 which is not valid bit width.
So a special syntax which supports it may be useful.

taichi-ishitani, 2024-03-08:
How about `$bitwidth` for the function name?

nblei, 2024-03-08:
Why should `logic<0>` be invalid?  Any variable or port of type `logic<0>` can be elided when transpiling to SV.

dalance, 2024-03-09:
If Veryl can evaluate the bit width, the `logic<0>` can be elided at transpile time.
But if the evaluation will be done at compile time of SystemVerilog, it can't.

For example, the following code is a parameterized selector.
If `N` becomes 1 by parmeter override, `S` becomes 0 and causes `sel: input logic<0>`.

```systemverilog
module Selector #(
  parameter  N: u32 = 2,
  localparam S: u32 = $clog2(N),
) (
  in : input  logic<N>,
  out: output logic   ,
  sel: input  logic<S>,
) {}
```

The module `Seletor` may be instantiated at SystemVerilog side, so Veryl can't elide the `logic<0>`.
If the instantiation is at Veryl side, the eliding is probably available by duplicating module definition to "Selector_with_sel" and "Selector_without_sel".
But if there are many elidable ports like "sel0, sel1,,,", the module duplication will be increased explosively.

By the above reasons, the workaround like `(W > 1) ? $clog2(W) : 1` is required, and I think it is nice that Veryl supports it.