veryl-lang/veryl #537: Repalce `parameter` and `localparam` keywords

taichi-ishitani, 2024-03-08:
The `parameter` and `localparam` keywords are relatively long keywords.
So I'd like to propose alternative words instead of these keywords.

* `param`
    * replacement of `parameter` keyword
* `const`
    * replacement of `localparam` keyword

I feel good appearance because these words have same length.

```systemverilog
module ModuleA #(
    param ParamA: u32 = 10,
    const ParamB: u32 = 10
)
```

nblei, 2024-03-08:
I am not a fan of this suggestion.

In many languages, `const` simply means that a variable cannot have its value modified.  A `localparam`, on the other hand, must be statically evaluable.

dalance, 2024-03-09:
In this context, `const` seems to mean "evaluated as constant value at compile time" like `constexpr` of C++.
But I agree the meaning of `const` is confusable because it is used through many languages by several meanings.

For example, `param` / `local` may be more familiar to SystemVerilog users?

```systemverilog
module ModuleA #(
    param ParamA: u32 = 10,
    local ParamB: u32 = 10
)
```

taichi-ishitani, 2024-03-10:
> param / local

I agree with this suggestion.

PR #549 (2024-03-12): Change parameter/localparam keyword to param/local
Closes #537 

### Migration guide

Change from `parameter` to `param`, and from `localparam` to `local`.

* From

```systemverilog
module #(
    parameter  ParamA: u32 = 1,
    localparam ParamA: u32 = 1,
) () {
}
```

* To

```systemverilog
module #(
    param ParamA: u32 = 1,
    local ParamA: u32 = 1,
) () {
}
```
closed 2024-03-12 by dalance