veryl-lang/veryl #564: [Feature] Integer type restricted to positive value

taichi-ishitani, 2024-03-15:
There are many cases that parameter values need to be restricted to positive value.
For example, bit width, FIFO depth.

I think integer type restricted to positive value is useful for these cases so I'd like veryl to suuprt such type.
SystemVerilog does not have such type so following code needs to be inserted to generated SV code.

```systemverilog
generate
  if (W < 0) begin : g_veryl_check_W
    $error("parameter W should be positive value");
  end
endgenerate
```

In addition, Pkl has a feature to add constraints to the given value.
I'd like Veryl to support such feature.
https://pkl-lang.org/main/current/language-reference/index.html#type-constraints

nblei, 2024-03-15:
I like this idea, and I think it fits more broadly into a discussion on what metadesign support should Veryl provide.

dalance, 2026-01-21:
In discussion with @taichi-ishitani, I feel `p8, p16, p32, p64` type seems to be good.
By #1847, Veryl compiler can detect whether values of these positive types become zero.
So assertion based checking is not necessary.