veryl-lang/veryl #567: [Feature] Iterate multi demension array

taichi-ishitani, 2024-03-16:
SystemVerilog provides foreach statement to iterate an array and the `foreach` statement can iterate a single dimension array only but also a multi dimension array.

To support this feature on Veryl, I'd like to extend `for` statement to take multiple iterator.

```systemverilog
var a: logic<10, 8>;
var b: logic<10, 8>

always_comb {
  for i: u32, j: u32 in 0..10, 0..8 {
    b[i][j] = ~a[i][j];
  }
}
```

nblei, 2024-03-16:
This is a nice idea.  I think veryl should also support iterators (and type inference), so in addition to what you have listed:

```veryl
always_comb {
  for (aa, bb) in a.iter().zip(b.iter()) { // zip implicitly checks that a.len() == b.len()
    // By type inference, we have (aa, bb) : (logic<8>, logic<8>)
    for (aaa, bbb) in aa.iter().zip(b.iter()) {
      // (aaa, bbb) : (logic, logic)
      bbb = aaa;
    } 
  }
}
```

This way, there are no accidental mistakes where, for example, the dimensions of `a` and `b` change, but the resulting procedural block is unmodified.

taichi-ishitani, 2024-03-27:
SystemVerilog doens not support `non local exits` feature.
Therefore, `multi demension for` loop needs to be convert one `for` loop if Veryl will also support `break` feature.

There are two way fot this.

* use multiple iterators

```systemverilog
int a[2][3];
for (int i = 0, int j = 0; (i < 2) && (j < 3); i += ((j == 2) ? 1 : 0), j += ((j == 2) ? -2 : 1)) begin
  $display("a[%0d][%0d] = %0d", i, j, a[i][j]);
end
```

I've confirmed that three major simulators support this notation but I'm not sure synthesis tools do.

* use single iterator

```systemverilog
int a[2][3];
for (int i_j = 0; i_j < (2 * 3); i_j++) begin
  automatic int i = i_j / 3;
  automatic int j = i_j % 3;
  $display("a[%0d][%0d] = %0d", i, j, a[i][j]);
end
```

taichi-ishitani, 2024-03-30:
Which syntex do you prefer?

```systemverilog
for i: u32 in 0..10, j: u32 in 0..20 {
 ...
}
```

```systemverilog
for i: u32, j: u32 in 0..10, 0..20 {
 ...
}
```

dalance, 2026-10-30:
I believe the necessity of this feature warrants further discussion.

As noted in the initial description, the scope supported by #3588 is effectively equivalent to that of nested `for` loops.
However, the implementation in #3588 introduces new constraints, such as the inability to use a `step` clause, that do not exist for nested `for` loops; from a user's perspective, this seems to lack consistency.

Furthermore, nested `for` loops appear superior in terms of the readability of the generated SystemVerilog code.
Are there any specific use cases or advantages that outweigh these concerns?

taichi-ishitani, 2026-10-30:
For SystemVerilog, a multi demensional array can be iterated with `foreach` statement.
It can help to reduce nested loops.
I'd like to introduce it into Veryl.

However, `foreach` statement can iterate array only but not multi demensional loops and can be used as a statement but not `generate` construction. So my proposal is enhancing `for` sattement but not introducing `foreach` statement directly.

While it would be possible to support a `step` feature for multidimensional `for` loops, I omit this feature for the initial implementation to keep things simple.

dalance, 2026-10-30:
I believe that constructs like `foreach`, as well as the iterators proposed by nblei, offer the advantage of preventing errors related to loop iteration counts.

However, simply reducing the nesting depth of a double loop by one level seems to offer almost no real benefit, other than saving a few characters.
While the number of nested levels or characters saved would increase in cases involving, say, five or ten nested loops,
such scenarios are extremely rare, and in those instances, the `for` statement itself would likely become excessively long.

Furthermore, my primary concern relates to the resulting SystemVerilog code after expansion.
I find it quite difficult to decipher the meaning of the code below; what are your thoughts on this?

```systemverilog
for (int __i_j = 0; __i_j < 8 * 4; __i_j++) begin
    int i        ;
    int j        ;
    i         = __i_j / 4;
    j         = 4 - 1 - __i_j % 4;
    o_a[i][j] = ~i_a[i][j];
end
```