veryl-lang/veryl #568: if expression does not allow less than (or greater than)

nblei, 2024-03-16:
For example:

```veryl
assign a   = if c < d   { b } else { c };
assign aa = if c <= d { b } else { c }; 
```

Second line is fine, but first line produces a ParserError::SyntaxError -- Unexpected token: LAngle.

Work around for now is to use `if c<= d && c != d`.

taichi-ishitani, 2024-03-16:
less than operator is >:

dalance, 2024-03-16:
This seems to be common error because the operators are differ from other languages. So I think a special error message which suggests  <: and >: is required.

dalance, 2024-03-21:
I'll change the error message like below:

```
Error: ParserError::SyntaxError

  × Unexpected token: '<', do you meant "less than operator '<:'" ?
    ╭─[/home/hatta/work/repos/veryl/testcases/veryl/03_operator.veryl:44:1]
 44 │     // binary compare
 45 │     assign a          = 1 < 1;
    ·                           ┬
    ·                           ╰── Error location
 46 │     assign aa         = 1 <= 1;
    ╰────
  help:
```

nblei, 2024-03-21:
Awesome.

Go with "do you mean", though.

PR #581 (2024-03-21): Improve parse error message
Closes #568
closed 2024-03-21 by dalance