veryl-lang/veryl #571: [Feature] '?' character in number literal

taichi-ishitani, 2024-03-18:
In systemVerilog, `?` character is a member of z digit so we can use it in a number literal.

```systemverilog
a = b inside {3'b??1};
```

I think `?` character shows `don't care` more strongly than `x` or `z` charactors.
Therefore, I'd like to use `?` character in a number literal but Veryl does not support it.

nblei, 2024-03-18:
What section in SV LRM describes`?`?

taichi-ishitani, 2024-03-18:
> What section in SV LRM describes`?`?

See '5.7.1 Integer literal constants' section.
[image]

nblei, 2024-03-18:
Thanks.

In this case, I would argue that Veryl is making the right decision by not using `?`.  Intuitively, a question mark is used to represent the presence of doubt or uncertainty.  This is quite different from `z`, used to indicate high impedance of a tristate buffer, and I think, is closer to representing an `x` value.  As such, I think `?` should not be used --- to me it seems to be used in SV literals in a very unintuitive way.

But perhaps I am missing something about quad-state logic:

Are the truth tables for `a inside {3'b1zz1}` and `a inside {3'b1xx1}` different?

taichi-ishitani, 2024-03-18:
`?` characther is used to emphasis "don't care", so I think it is little different from `z` and also `x` values. Therefore, I'd like to import `?` character from SV.

> Are the truth tables for a inside {3'b1zz1} and a inside {3'b1xx1} different?

In the `inside` operator `x`/`z`/`?` is treated as "don't care" condition. So these expressions has the same result.

> My assumption is that they are both equivalent to a[3] & a[0].

Yes your understanding is correct.

nblei, 2024-03-18:
> `?` characther is used to emphasis "don't care", so I think it is different from `z` and also `x` values. Therefore, I'd like to import `?` character from SV.

But the LRM specifically states that `?` is an alternative for `z`.

taichi-ishitani, 2024-03-18:
>  the LRM specifically states that ? is an alternative for z.

Yes, `?` is an alternative for `z` and also more readable version of `z`.

> it is in others such as a casez.

SV has the `case inside` statement. This is enhanced version of `casez` statements.
So I think 'case inside` should be imported from SV instead of `casez`, and the custom that `z` is used as "don't care" condtion should be not inherited.

dalance, 2024-03-19:
I think it is not good that there are the three type representations of don't care.
The meaning of `?zx` seems to be bit difference respectively from don't care, if anyone should be selected, I think `x` may be better.
Thiis is because `x` seems to be the most common representation of don't care.

https://en.wikipedia.org/wiki/Don%27t-care_term#X_value

taichi-ishitani, 2024-03-19:
I understood. So I'm closing this issue.

closed 2024-03-19 by taichi-ishitani