veryl-lang/veryl #579: Configuration property to disable adding poject prefix to generated SV files

taichi-ishitani, 2024-03-20:
I think it is useful to disable adding project prefix to generated SV files in the case that rewriting existing SV modules by using Veryl.
Therefore, I'd like to add a new configuration property to control adding project prefix to generated SV files.

dalance, 2024-03-21:
I think there are two approaches.
How about it?

* build property `omit_projectname`
If the property is set to `true`, the project prefix will be removed from all items of the project.

```toml
[build]
omit_projectname = true
```

* specify generating name by attribute

```systemverilog
#[export(ModuleX)]
module ModuleA {
}
```

taichi-ishitani, 2024-03-21:
> * build property `omit_projectname`
>   If the property is set to `true`, the project prefix will be removed from all items of the project.
> 
> ```toml
> [build]
> omit_projectname = true
> ```

I prefer the first one.

taichi-ishitani, 2024-03-25:
I tried to use this configuration property but I faced a problem.
Generated module/package names have no project prefix but identifiers to refer modules/packages includes the project prefix yet.

```systemverilog
module pzbcm_ram
  import pzbcm_veryl_pzbcm_common_pkg::*;
 #(
  parameter int unsigned WORD_SIZE     = 8                       ,
  parameter int unsigned ADDRESS_WIDTH = calc_width(WORD_SIZE, 1),
  parameter int unsigned DATA_WIDTH    = 8                       ,
  parameter type         DATA_TYPE     = logic [DATA_WIDTH-1:0]  ,
  parameter bit          BUFFER_OUT    = 0                       ,
  parameter bit          USE_RESET     = 0                       ,
  parameter DATA_TYPE    INITIAL_VALUE = DATA_TYPE            '(0)
) (
```

PR #582 (2024-03-21): Add omit_project_prefix option
Closes #579
PR #592 (2024-03-25): Fix omit_project_prefix of file scope import
Closes #579
closed 2024-03-25 by dalance