veryl-lang/veryl #580: Generate SV code without comments

saturn77, 2024-03-20:
The ability to generate documentation from the veryl code is extremely useful. 

It would also be helpful to have an option to remove or strip away the comments that are in the veryl code when generating the SystemVerilog code. In cases where the veryl code is heavily documented, this would allow inspection of the SystemVerilog code in a plain format and make it easier to review the generated code.

This process to do this would simply be placing a "strip_comments"  in the toml / veryl file as: 

```toml 
[format]
indent_width = 4
strip_comments = true 

```

dalance, 2024-03-21:
Looks good.
`build` may be more suitable because it relates code generation.

```toml
[build]
strip_comments = true
```

saturn77, 2024-03-21:
Yes, sounds good. Having the format as below makes more sense.
```toml
[build]
strip_comments = true
```
I also really like how the actual formatted code in the SystemVerilog mirrors the code formatting in Veryl. The only exception is in the case statement, but I can live with the way that it is now. 

It would be nice to have an option, however. 

For example 

```toml 
[build]
sv_case = column_case
```
Which would make this output code: 
```systemverilog 
    always_ff @ (posedge clock) begin
        case (state)
            rx_IDLE : sample_counter <= CLOCKS_PER_BAUD / 2;
            rx_SHIFT: if ((sample_counter > 0)) begin
                sample_counter <= sample_counter - 1;
            end else begin
                sample_counter <= CLOCKS_PER_BAUD - 1;
            end
            rx_LOAD: sample_counter <= 0;
            rx_DONE: sample_counter <= 0;
        endcase
    end
```

being transformed to this output code instead: (what I call column case)

```systemverilog 
    always_ff @ (posedge clock) begin
        case (state)
            rx_IDLE :  sample_counter <= CLOCKS_PER_BAUD / 2;
            rx_SHIFT:  if ((sample_counter > 0)) begin
                           sample_counter <= sample_counter - 1;
                       end else begin
                           sample_counter <= CLOCKS_PER_BAUD - 1;
                       end
            rx_LOAD:   sample_counter <= 0;
            rx_DONE:   sample_counter <= 0;
        endcase
    end
```

dalance, 2024-03-21:
The column_case seems to be good too.
I think it may be better that it is by default for both Veryl and generated SV.

PR #585 (2024-03-21): Add strip_comments option
Closes #580
closed 2024-03-21 by dalance