veryl-lang/veryl #595: Veryl std import

saturn77, 2024-03-26:
Considering the Veryl std lib that is in progress, it would be nice to have a Veryl directive to import the standard library
with some syntax that indicates that it is being loaded from the git repository. 

Additionally, for any user that develops their own standard library, they could import in the same way. 

I like this method versus declaring the std library import in the toml file, as it is more distinctive. 

An idea is shown below.

```systemverilog

const std = @import("https://github.com/veryl-lang/std");

```

The syntax is somewhat borrowed from Zig, versus Rust. I like the idea that a "const" is assigned to a library import, and then different  methods (modules) could be referenced from the "std" object.

So considering the instantiation of a module from the standard library: 

```systemverilog
inst u1_counter : std.__counter__ (
        i_clk,
        i_rst, 
        i_clear, 
        i_set, 
        i_set_value ,
        i_up, 
        i_down, 
        o_count, 
        o_wrap_around,  
    );

```

dalance, 2024-03-28:
I think this style causes difficulty for dependency and version resolver.
TOML can be read before reading source code, so resolver can fetch dependencies, generate lock file and decide which files should be parsed all at once.
But in this style, each `import` will trigger resolver incrementally.

How does Zig resolve this issue?

saturn77, 2024-03-29:
@dalance I went and took another look at how the TOML file handles dependencies, and I understand now that this functionality  already exists with TOML with [dependencies]. 

Also I am not sure how Zig handles the issue, but I like the concept and syntax of their imports. Perhaps their is some internal cache in the compiler that doesn't have to resolve/fetch the dependencies each time their is a compile of the main file. 

I think this issue can be closed. Thanks, James

dalance, 2024-03-29:
OK. I'll close this issue.

closed 2024-03-29 by dalance