veryl-lang/veryl #599: [[Feature] var declaration in block in function

taichi-ishitani, 2024-03-26:
Currently, Veryl does not support `var declaration` in block in function.
[image]

Short lifetime of variable is good for readable code so I'd like Veryl to support this feature.

saturn77, 2024-03-31:
@taichi-ishitani This is a good feature as described above. 

Also, in addition to var, there may be the options to include a **const**. 

I just started to review IEEE 1800-2023, which was released in February 2024. 

In 1800-2023, there is a section 6.20.6 which describes the new **const** keyword. It is something to be aware of for future Veryl support. 

```text 
A const form of constant differs from a localparam constant in that the localparam shall be set during
elaboration, whereas a const can be set during simulation, such as in an automatic task.
A static constant declared with the const keyword can be set to an expression of literals, parameters, local
parameters, genvars, enumerated names, a constant function of these, or other constants. Hierarchical names
are allowed because constants declared with the const keyword are calculated after elaboration.
```

I can see this being useful for cases where there are iterations on a module's parameters during simulation.

saturn77, 2024-03-31:
Another question or comment is related to static var type inside of a function.

For the code below, is FuncA "automatic" type , i.e. always re-entrant ? 

A suggestion for static function is shown in add_u8 below. 

```systemverilog
// package definition
package PackageA {
    local ParamA: u32 = 1;
    local ParamB: u32 = 1;

    function FuncA (
        a: input  logic<ParamA>,
    ) -> logic<ParamA> {
        return a + 1;
    }

    static function add_u8 ( logic<8> x, logic<8> b) -> logic<9> {
        static var z = 0;   // local static var, initialized to 0
        z = z + x  + y;
        return z; 
    }
}

module ModuleA {
    let a : logic<10> = PackageA::ParamA;
    let _b: logic<10> = PackageA::FuncA(a);
    
    let result : logic<9> = 0; 
    result = PackageA::add_u8(1,2);   
    
    $display("result = %d", result);  // prints 3 

    result = PackageA::add_u8(3,4); 
    $display("result = %d", result);  // prints 10 ??
    
}
```

taichi-ishitani, 2024-03-31:
> static var z = 0;   // local static var, initialized to 0

Design Compiler will ignore this kind of initialization.
Therefore, I'm not sure Veryld should support `static` lifetime.

saturn77, 2024-03-31:
AMD-Xilinx UG901 only support static functions, according to latest document 2023.

nblei, 2024-03-31:
Variables declared inside functions must be static, and functions must be static if we want them to be synthesizable.  Synthesizing reentrant functions is, in general, impossible.  I agree with @taichi-ishitani that `static` keyword is thus not needed in Veryl.

According to SV LRM 13.4.2, functions are implicitly given a static lifetime unless they are declared with the `automatic` keyword, or they are defined within a module, interface, program, or package which is declared as `automatic`.

Since `automatic` is not emitted by Veryl, then it means all emitted functions are static.  However, it does not hurt to explicitly mark the emitted SV functions as `static`.

taichi-ishitani, 2024-04-01:
> AMD-Xilinx UG901 only support static functions, according to latest document 2023.

[image]

Vivado supports both of static and autoamtic function.

saturn77, 2024-04-01:
I personally have never used a static function with Vivado, with my understanding that "static" functions will be treated as automatic per UG901.

@taichi-ishitani The reference that you make is at the end of UG901, and is in conflict with the the descriptions earlier in the document. There are many forum articles on amd-xilinx about this. 

Basically, I do not use static functions for synthesis with vivado.

In the context of other synthesis tools or asic tools I am not sure how they are treated.

I have always used automatic for synthesis as an example below. 

```systemverilog
    //------------------------------------------------------
    // Local Functions 
    //------------------------------------------------------
    function automatic int unsigned accumulate_with_limit (int unsigned te_count);
	    if (te_count < TE_MAX_COUNT) begin
		    te_count = te_count + 1;
	    end else begin
	        te_count = TE_MAX_COUNT;
	    end
	    accumulate_with_limit = te_count;
    endfunction

    function automatic int unsigned hold(); 
	    hold = 1; 
    endfunction
```

[image]

saturn77, 2024-04-01:
Vivado **synthesis**  treats all functions as automatic. 

[image]

taichi-ishitani, 2024-04-01:
> Since `automatic` is not emitted by Veryl, then it means all emitted functions are static.

Veryl emmitd all SV functions are `automatic` but not `static`.
See this testcase.
https://github.com/veryl-lang/veryl/blob/c34f5cd377f08f7a79dc41f33b2eaffabc1835ef/testcases/sv/06_function.sv#L5

nblei, 2024-04-01:
My bad, I hope I didn't add confusion with my horrible and wrong description of all functions needing to be static, when of course, `automatic` allocation leads to static-hardware allocation.  I was confusing `automatic` with re-entrant.

closed 2024-09-02 by dalance