veryl-lang/veryl #603: Cannot find member variable from struct input port

taichi-ishitani, 2024-03-27:
Veryl reports an error unexpectedly for member reference of struct input port.
You can repdoduce this error by building following code.

```systemverilog
package foo_bar_baz_pkg {
  struct foo_bar_baz_type{
    foo: logic,
    bar: logic,
    baz: logic
  }
}

module foo_bar_baz (
  i_foo_bar_baz: input foo_bar_baz_pkg::foo_bar_baz_type,
  o_foo: output logic,
  o_bar: output logic,
  o_baz: output logic
){
  assign o_foo = i_foo_bar_baz.foo;
  assign o_bar = i_foo_bar_baz.bar;
  assign o_baz = i_foo_bar_baz.baz;
}
```

[image]

dalance, 2024-03-28:
This seems to be resolved at master branch.

closed 2024-03-28 by dalance

taichi-ishitani, 2024-03-29:
Thank you!
I've confirmed this issue is resolved.