veryl-lang/veryl #619: Output file combination

saturn77, 2024-03-31:
This is a **feature** suggestion for the build tool.

The suggestion is to add the option to bundle or pickle all systemverilog output files into a single file.  

Some tools, like the morty tool have done this (https://github.com/pulp-platform/morty). I would like to suggest adding the similar feature to the [build] section of the toml file. 

For example, 

```toml
[build]
strip_comments = true
bundle = true
top_level = "uart_top_level.sv"
```
The above example would strip comments and put all source files into a single file name, and the "top_level" would designate the file name for the top level file. If one left the top_level file name blank, then it would use the proj_name as the top level file. 

I think this would be a great feature for handling large designs as well as creating re-usable IP, for example, say someone made a project with 

- uart_tx.veryl
- uart_rx.veryl
- uart_parser.veryl
- uart_fifo.veryl
- uart_package.veryl

Then with 

```toml
[build]
strip_comments = true
bundle = true
top_level = "top_level_uart_system.sv"
```
all five files would be bundled into top_level_uart_system.sv.

nblei, 2024-03-31:
I do not see much utility in this feature, since veryl already produces a file `<project-name>.f` which contains all the generated sverilog files.

So something as simple as `cat $(cat uart.f) >top_level_uart_system.sv` would generate the desired file.

I also don't think veryl should be **promoting** bad practices, such as placing all modules into a single file.

saturn77, 2024-03-31:
I cannot see a logical reason why it is a bad practice to have a single SystemVerilog file, when the design language is Veryl.  

I have written several EDA tools, widely used within companies, particularly for very large logic analyzer designs among other things. Having a single file is advantageous in an automated tool chain where everything is combined into one file and the compiler is kicked off with this single file. That is why pulp platform wrote morty, and they are not the first to write such a tool. 

The general approach that is the foundation here is :  

1. Design mode == individual modules each have their own source file (Veryl)
2. Compilation mode == combined modules into a single file (systemverilog)

Source code is read more often that it is written, so for readability purpose, I would not recommend combining multiple modules into a signal file for the source of truth root file (Design mode), which is in the case Veryl. 

But part of Veryl's advantage is that is also is an integrated build tool in combination with being a compiler.  

Hence generating a single file is simply part of Compilation mode option.

dalance, 2024-04-05:
There are two use cases that come to mind for outputting a single file. One is due to constraints of EDA tools; for example, about 20 years ago, in our project, we used `include` statements to read the entire project from a single file because specific EDA tools became very unstable when reading many files. Another is when providing the project to others. It is convenient to have a single file to prevent troubles related to the order and format of the file list when providing it as an IP to customers or to EDA vendors for support. In this case, readability of the output is not a concern and it is often encrypted.

Considering that one of Veryl's goals is to be integrated into existing projects, I think there is a reason to support projects in a single file format. Moreover, among the already supported multiple file list formats, adding one more output format would not particularly endorse a specific approach.

dalance, 2024-04-16:
I added `bundle` target.

```toml
[build] 
target = {type = "bundle", path = "output.sv"}
```

PR #671 (2024-04-16): Add bundle mode to target config
Closes #619
closed 2024-04-16 by dalance