veryl-lang/veryl #624: SDC support

dalance, 2024-04-01:
If Veryl can identify clock and reset by #622, SDC support like auto generation and consistency check can be added.

nblei, 2024-06-04:
I think there's a very powerful idea here, Naoya.

Rather than writing synthesis scripts in some typeless scripting language which has no understanding of the Verilog modules being synthesized, doesn't it make sense to write synthesis scripts in a language with tooling which can parse and extract meaning from the HDL?

In other words, can we leverage the LSP and semantic checks of Veryl to provide a powerful DSL which transpiles into tcl synthesis scripts?

dalance, 2024-06-05:
I thought handling SDC directly, but a specialized language which can be transpiled into TCL may be better surely.

taichi-ishitani, 2024-09-01:
See also:
https://accellera.org/activities/working-groups/clock-domain-crossing
https://vimeo.com/942927981?&signup=true#_=_