veryl-lang/veryl #626: Cannot use `typdef` inside interface

taichi-ishitani, 2024-04-02:
Veryl reports an error unexpectedly for `typedef` inside interface declaration.

[image]

nblei, 2024-04-02:
Annex A.2.9 of IEEE 1800-2023 (IEEE don't sue me please!):

```
A.2.9 Interface declarations
modport_declaration ::= modport modport_item { , modport_item } ;
modport_item ::= modport_identifier ( modport_ports_declaration { , modport_ports_declaration } )
modport_ports_declaration ::= { attribute_instance } modport_simple_ports_declaration
                                                     | { attribute_instance } modport_tf_ports_declaration
                                                     | { attribute_instance } modport_clocking_declaration
modport_clocking_declaration ::= clocking clocking_identifier
modport_simple_ports_declaration ::= port_direction modport_simple_port { , modport_simple_port }
modport_simple_port ::= port_identifier
                                           | . port_identifier ( [ expression ] )
modport_tf_ports_declaration ::= import_export modport_tf_port { , modport_tf_port }
modport_tf_port ::= method_prototype
                                 | tf_identifier
import_export ::= import | export
```

I don't think you can actually declare a type inside an interface.  I think you have to declare the type inside a package, and then import the type from that package.

taichi-ishitani, 2024-04-02:
`typedef` can be used inside an interface declaration.
This is the syntax tree from interface declaration to type declaration.

* interface_declaration
    * interface_item
        * non_port_interface_item
            * interface_or_generate_item
                * module_common_item
                    * module_or_generate_item_declaration
                        * package_or_generate_item_declaration
                            * data_declaration
                                * type_declaration

closed 2024-04-05 by dalance