veryl-lang/veryl #628: Incorrect behavior for hierarchical identifiers for nested structures within modports

anantajit, 2024-04-02:
Note: this is a similar issue to #541

I'm running Veryl v0.8.2, with the following design:

```
  package pkg {
  struct DataStruct {
    test: logic<32>,
  }
}

interface X {
    var data: pkg::DataStruct;

    modport port {
      data: input
    }
}

module Y (
  dataport: modport X::port
) {

  assign dataport.data.test = 32'h23;

}

```

This is met with the error `Error: unknown_member (link)`, which appears to be a bug. Note that the snippet below passes `veryl check`:

```
package pkg {
  struct DataStruct {
    test: logic<32>,
  }
}

interface X {
    var data: pkg::DataStruct;

    modport port {
      data: input
    }
}

module Y (
  dataport: modport X::port
) {

  var datav: pkg::DataStruct;
  assign datav.test = 32'h23;

  assign dataport.data = datav;

}
```

nblei, 2024-04-02:
@dalance I think it is likely that there are many corner cases involved in the whole namespace/symbol-table/scoping resolution stuff.  Perhaps it may be time to rethink the symbol table design?

dalance, 2024-04-05:
@nblei Thank you for your suggestion. I found many obsolete code related to SystemVerilog namespace, so I'll refactor them and fix this issue at the same time.

PR #631 (2024-04-06): Fix resolve failure for some nested items
Closes #628
closed 2024-04-06 by dalance