veryl-lang/veryl #633: [FEATURE] In-line Struct Assignment

anantajit, 2024-04-06:
As I understand the current Veryl documentation, struct assignment in Veryl cannot be done in-line and each field must be initialized separately:

```
struct X {
  field_1: ...,
  field_2: ...
}

...

var x: X;
x.field_1 = ...
x.field_2 = ...
```

I think that the following Rust-like syntax would be best for this use-case: 
```
let x = X { field_1: ..., field_2: ... };
```

dalance, 2025-03-21:
I'll add this feature because `const` assignment can't be splitted to field assignments.

```systemverilog
struct Param {
    a: bit<10>,
    b: bit<10>,
}

const p: Param = Param {
    a: 10,
    b: 10,
};

const q: Param = Param {
    a: 1,
    ..default(0), // as the same as `..Default::default()` in Rust
};
```

This will be transpiled into below:

```systemverilog
localparam Param p = '{
    a: 10,
    b: 10
};

localparam Param q = '{
    a: 1,
    default: 0
};
```

`const p: Param = Param {}` may be bit redundant, but it may be simplified like `const p = Param {}` by type inferation in future.

dalance, 2025-03-21:
`{` after expression is used by many syntaxes like `if expression {`.
So `Param {` is ambiguous. `Param '{` can be used instead of it.

```systemverilog
struct Param {
    a: bit<10>,
    b: bit<10>,
}

const p: Param = Param '{
    a: 10,
    b: 10,
};

const q: Param = Param '{
    a: 1,
    ..default(0), // as the same as `..Default::default()` in Rust
};
```

closed 2025-03-24 by dalance